ai-republic/bms-to-inverter #92: Support and Testing of Soluna Battery EOS 5k Pack

srobison, 2025-01-24:
Hi Torsten!
First of all, thank you so much for developing this tool - so useful! I've recently acquired several [Soluna EOS 5K](https://soluna.co/wp-content/uploads/2024/12/EOS-5K.pdf) battery packs that will eventually be used with a hybrid LV inverter.  I needed a device that could communicate with them to verify they are still good and to keep the battery enabled long enough to recharge after sitting for an extended period.  After following your How-to-use instructions, I was able to get my raspi 4B + [Waveshare 2 CH CAN FD hat](https://www.waveshare.com/wiki/2-CH_CAN_FD_HAT) communicating using the `PYLON_CAN` BMS profile in your configuration utility.
The good news I can now monitor the status and keep the pack enabled to recharge!
The less good news is that not all of the parameters seem to be updating.  I do read the top 3 important ones for the pack: voltage, current, and SoC.  However I don't see any cell info, and there are apparent reports of a few alarm conditions.  Also, the values seem very slow (a couple minutes between changes) to update to the web server, though I can't say if that's the fault of the battery.  Polling is set to 1s. 

[image]

Please let me know if you have suggestions for improving the communications.
Additionally, if there is anything I can do to help test this setup for your project I would be happy to help.

BR

Sterling

ai-republic, 2025-01-25:
Hi @srobison glad you like the application 😄 
Unfortunately the Pylon CAN communication specification does not support single cell info. So there is nothing I can do about that.
The only option would be to use the Pylon RS485 specification which supports also single cell values. But for the current _PYLON_RS485_ binding that's not yet fully implemented. Might do that tomorrow actually 😉 
The slowness of your value updates should not be the webserver as it polls the data every second. The application also is polling with 1 sec interval so that means your BMS is updating those slowly.

srobison, 2025-02-03:
Thanks for the clarifications @ai-republic ! I'll check back here periodically for the full support `PYLON_RS485` and give that a shot when ready.

closed 2025-02-03 by srobison

ai-republic, 2025-02-08:
@srobison I'll let you know once I implemented it