ai-republic/bms-to-inverter #93: Bms monitored with batmon and ha

erlendoyen, 2025-01-27:
Hi

I am already nontoring 2 bms's with batmon on ha via Bluetooth.

Batmon is already using mqtt to stroke data.
Is there anyway of getting this application to be able to either use the device running the device running ha core as a bt proxy to retrive data from the bms's or some other way to get data from the existing mqtt instance

End goal is to feed data from the 2 bms's at the same time as feeding data via can to a epever up hi inverter.

ai-republic, 2025-01-29:
Hi @erlendoyen,
first of all, how do you publish the Batmon data to your MQTT broker? Is that via bluetooth? What MQTT broker are you using?
In theory I could connect to the existing MQTT broker topic and retrieve the data there. Another option would be to connect your BMS to a RPI via CAN, RS485 or ModBus and then read that data and publish it to the MQTT from which HA can receive the data. This way is more or less implemented (needs only the exact Json message that needs to be sent to HA).

erlendoyen, 2025-01-29:
i use a rbpi with ha>>bluetooth to the bms'es
and mqtt is running on the same rbpi in ha

ai-republic, 2025-01-29:
Is the Batmon connected via _WLAN / RJ45_ or via _bluetooth_ to your RPi and publishes data to the MQTT broker on your PI?
And what MQTT broker, i.e. Mosquitto(?) do you have installed on your RPi?
In theory I could connect to the existing MQTT broker topic and retrieve the data there. I could create a virtual MQTT port that would need the broker IP, credentials and topic so that I can read the battery data from there.

erlendoyen, 2025-01-29:
batmon is running on the rpi runna ha, Mosquitto mqtt is also running on the same rpi running ha.
and batmon is connected to the bms directly from the rpi running ha via bluetooth.
i am running HA core (os)

erlendoyen, 2025-02-13:
Hi

Any updates or ways to add the jbd Bluetooth bms to this solution?

ai-republic, 2025-02-15:
No, I have been really busy. I'll try to come up with something tomorrow

erlendoyen, 2025-02-15:
"Hi, that's okay, no pressure. I was just curious. This is a nice project, but non-commercial, so I understand that features are not implemented that fast or upon request every time."

closed 2026-01-30 by ai-republic