ai-republic/bms-to-inverter #95: Bug found - Samsung BMS

chuotde, 2025-01-31:
My setup:

- Inverter:  Solis S5-EH1P5K-L
- Battery: 2x Samsung ELPM482-00005 in parallel
- RPi with 2-CAN HAT, can0: BMS, can1: inverter
- BMS-to-Inverter configuration: can0: SAMSUNG_CAN; can1: PYLON_CAN

Issue: Discharge current limit sent to inverter is wrong, inverter sees it as 645.9A

can0 dump: can0  502   [8]  45 02 AC 03 **AC 03** C0 01  --> charge and discharge limit are the same 0x3AC
can1 dump: can1  351   [8]  45 02 AC 03 **54 FC** C0 01 --> discharge limit is 0xFC54 (64596)

Bug found in: SamsungBmsCANProcessor.java, line 151

`// Discharge current limit (0.1A) - uint_16
 pack.maxPackDischargeCurrent = data.getChar() *** -1**;`   --> should not times the value by -1, (0x3AC + 0xFC54 = 0x10000)

Can you please fix it? Many thanks.

chuotde, 2025-02-01:
Maybe another bug with cells voltage display on webUI: only 6 cells are available out of 28 cells

[https://ibb.co/jsSc7bF](url)

I propose a change like this:

`               case 0x5F0:
                    readCellVoltages(pack, data, 0);
                break;
                case 0x5F1:
                    readCellVoltages(pack, data, 3);
                break;
                case 0x5F2:
                    readCellVoltages(pack, data, 6);
                break;
                case 0x5F3:
                    readCellVoltages(pack, data, 9);
                break;
                case 0x5F4:
                    readCellVoltages(pack, data, 12);
                break;`

`protected void readCellVoltages(final BatteryPack pack, final ByteBuffer data, final int cellNoStart) {
        // Tray id
        final int tray = data.getShort() - 1;
        // Cell voltage (1mV)
        pack.cellVmV[cellNoStart + tray * 14] = data.getChar();
        // Cell voltage (1mV)
        pack.cellVmV[cellNoStart + tray * 14 + 1] = data.getChar();

        if (cellNoStart != 12) {
            // Cell voltage (1mV)
            pack.cellVmV[cellNoStart + tray * 14 + 2] = data.getChar();
        }
    }`

ai-republic, 2025-02-02:
Hi @chuotde, thanks for the bug spotting :)
I've added the changes. Please re-test (_Clean install_) with the latest changes and let me know if it works 😄

chuotde, 2025-02-02:
Hi,

- 1st issue with wrong max discharge current: no change (I still see * -1 in your code - line 151)
- 2nd issue with cells voltage: all look good. 

Thanks for your quick fix

ai-republic, 2025-02-08:
@chuotde thanks for the feed back. I fixed the discharge also. Could you please retest and confirm?

chuotde, 2025-02-13:
Hi @ai-republic, 

Thank you for your support, discharge limit is now correct. You can now close the issue.

closed 2025-02-15 by ai-republic