ai-republic/bms-to-inverter #96: Unsupported BMS - Shoto Smartli

hpeyerl, 2025-02-01:
I have 8 x [Shoto Smart-Li SDA10-48100 batteries](https://www.shuangdeng.com.cn/view-3020-20003.html) currently supported by Solar-Assistant via rs485.

I played with bms-to-inverter and have it running on a Pi3b with CAN+RS485 hat, and have tried a number of BMS protocols (Narada, Huawei, etc) and it appears none of them are able to talk to these batteries.   

I browsed through the various bms-* directories looking at modbus registers and none of them appear to match the documentation for my batteries.

I have a pdf that describes the modbus protocol and a registers definition file for the [modbus-cli](https://github.com/favalex/modbus-cli) tool which I've successfully used to read every register.  

[Shoto RS485 Modbus protocol V1.2.pdf](https://github.com/user-attachments/files/18629937/Shoto.RS485.Modbus.protocol.V1.2.pdf)

[registers.txt](https://github.com/user-attachments/files/18629926/registers.txt)

My goal is for my Solark 15k to be able to get accurate information from these batteries via CAN bus.

What can I do to help?

ai-republic, 2025-02-02:
Hi @hpeyerl,
I will have a look at the specification and possibly see that I can implement an initial binding by next weekend.

hpeyerl, 2025-02-02:
Here's a register dump from the battery at Addr=2. 

[shoto-2.txt](https://github.com/user-attachments/files/18632420/shoto-2.txt)

hpeyerl, 2025-02-04:
I fork'd the repo and started adapting some code.  In case it helps.

https://github.com/hpeyerl/bms-to-inverter/tree/main/bms-shoto-smartli-modbus/src/main/java/com/airepublic/bmstoinverter/bms/shotosmartli/modbus

We'll see how far I get by the time you get around to it.

ai-republic, 2025-02-09:
Hi @hpeyerl,
I've implemented the Shoto BMS binding _SHOTO_MODBUS_. Please download the updated _Configurator_ app and then do a _Clean install_. Also enable debug log-level and inverter binding _NONE_ just to check the BMS readings. Please let me know how it goes and attach the log file if you encounter issues

hpeyerl, 2025-02-09:
Thanks.  I've tried it quickly and it's not able to read from the BMS.

My well-pump failed so I'm a bit pre-occupied with that at the moment.  I'll try to dig in further later.

The first pack address is 2.  Looks like it's trying '0'.  

```
2025-02-09 16:27:05.372 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: config/config.properties
2025-02-09 16:27:08.655 | INFO  | main       | ore.InverterProducer:76    | Created inverter binding: SOLARK_CAN
2025-02-09 16:27:08.801 | INFO  | main       | verter.core.Inverter:53    | Using plugins: []
2025-02-09 16:27:08.839 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #0[SHOTO_MODBUS] on port /dev/ttyUSB0
2025-02-09 16:27:09.681 | INFO  | main       | .webserver.WebServer:203   | Starting webserver on ports 8080:8443
2025-02-09 16:27:09.701 | INFO  | main       | .jetty.server.Server:536   | jetty-12.0.12; built: 2024-07-25T21:58:37.668Z; git: cc6f1b74db755fed228b50701ad967aeaa68e83f; jvm 17.0.13+11-Raspbian-2deb12u1rpt1
2025-02-09 16:27:09.830 | INFO  | main       | aultSessionIdManager:328   | Session workerName=node0
2025-02-09 16:27:09.842 | WARN  | main       | dler.ResourceHandler:97    | Base Resource should not be an alias
2025-02-09 16:27:09.903 | INFO  | main       | er.AbstractConnector:326   | Started ServerConnector@1baf3{HTTP/1.1, (http/1.1)}{0.0.0.0:8080}
2025-02-09 16:27:11.125 | INFO  | main       | sl.SslContextFactory:335   | x509=X509@8f9c2e(sitename,h=[torsten oltmanns],a=[],w=[]) for Server@d1230[provider=null,keyStore=jar:file:/home/hpeyerl/Downloads/bms_inverter/lib/webserver-0.0.1-SNAPSHOT.jar!/ssl/keystore.jks,trustStore=null]
2025-02-09 16:27:11.756 | INFO  | main       | er.AbstractConnector:326   | Started ServerConnector@10ea6b8{SSL, (ssl, http/1.1)}{0.0.0.0:8443}
2025-02-09 16:27:11.766 | INFO  | main       | .jetty.server.Server:593   | Started oejs.Server@12f4a93{STARTING}[12.0.12,sto=0] @12250ms
2025-02-09 16:27:11.768 | INFO  | main       | .webserver.WebServer:206   | Started webserver on ports 8080:8443 successfully!
2025-02-09 16:27:11.786 | INFO  | main       | verter.BmsToInverter:264   | Starting BMS receiver...
2025-02-09 16:27:11.796 | INFO  | Thread-11  | verter.BmsToInverter:290   | Reading BMS #0 SHOTO_MODBUS on /dev/ttyUSB0...
2025-02-09 16:27:11.800 | INFO  | Thread-11  | toinverter.core.Port:111   | Opening /dev/ttyUSB0 ...
2025-02-09 16:27:11.877 | INFO  | Thread-11  | toinverter.core.Port:113   | Opening port /dev/ttyUSB0 SUCCESSFUL
2025-02-09 16:27:30.391 | ERROR | Thread-11  | toBmsModbusProcessor:46    | Error reading from modbus!
java.io.IOException: Error reading from modbus device #0(address: 30001, numRegisters: 1)
	at com.airepublic.bmstoinverter.protocol.modbus.J2ModMasterPort.readHoldingRegisters(J2ModMasterPort.java:182) ~[protocol-modbus-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.protocol.modbus.J2ModMasterPort.sendFrame(J2ModMasterPort.java:124) ~[protocol-modbus-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.shoto.modbus.ShotoBmsModbusProcessor.sendMessage(ShotoBmsModbusProcessor.java:52) ~[bms-shoto-modbus-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.shoto.modbus.ShotoBmsModbusProcessor.collectData(ShotoBmsModbusProcessor.java:38) ~[bms-shoto-modbus-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$3(BmsToInverter.java:291) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
2025-02-09 16:27:30.420 | INFO  | main       | verter.BmsToInverter:312   | Starting inverter sender...
2025-02-09 16:27:30.420 | INFO  | Thread-11  | verter.BmsToInverter:342   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff

```

hpeyerl, 2025-02-09:
After reading the instructions, I set the 'id' to '2' in the configurator and now it errors out immediately after trying to read.

```
2025-02-09 17:38:52.853 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: config/config.properties
2025-02-09 17:38:56.072 | INFO  | main       | ore.InverterProducer:76    | Created inverter binding: SOLARK_CAN
2025-02-09 17:38:56.219 | INFO  | main       | verter.core.Inverter:53    | Using plugins: []
2025-02-09 17:38:56.255 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #2[SHOTO_MODBUS] on port /dev/ttyUSB0
2025-02-09 17:38:57.108 | INFO  | main       | .webserver.WebServer:203   | Starting webserver on ports 8080:8443
2025-02-09 17:38:57.128 | INFO  | main       | .jetty.server.Server:536   | jetty-12.0.12; built: 2024-07-25T21:58:37.668Z; git: cc6f1b74db755fed228b50701ad967aeaa68e83f; jvm 17.0.13+11-Raspbian-2deb12u1rpt1
2025-02-09 17:38:57.260 | INFO  | main       | aultSessionIdManager:328   | Session workerName=node0
2025-02-09 17:38:57.273 | WARN  | main       | dler.ResourceHandler:97    | Base Resource should not be an alias
2025-02-09 17:38:57.336 | INFO  | main       | er.AbstractConnector:326   | Started ServerConnector@fa3527{HTTP/1.1, (http/1.1)}{0.0.0.0:8080}
2025-02-09 17:38:58.563 | INFO  | main       | sl.SslContextFactory:335   | x509=X509@1eb5790(sitename,h=[torsten oltmanns],a=[],w=[]) for Server@e7c8c3[provider=null,keyStore=jar:file:/home/hpeyerl/Downloads/bms_inverter/lib/webserver-0.0.1-SNAPSHOT.jar!/ssl/keystore.jks,trustStore=null]
2025-02-09 17:38:59.180 | INFO  | main       | er.AbstractConnector:326   | Started ServerConnector@e48808{SSL, (ssl, http/1.1)}{0.0.0.0:8443}
2025-02-09 17:38:59.190 | INFO  | main       | .jetty.server.Server:593   | Started oejs.Server@aee2a2{STARTING}[12.0.12,sto=0] @10112ms
2025-02-09 17:38:59.192 | INFO  | main       | .webserver.WebServer:206   | Started webserver on ports 8080:8443 successfully!
2025-02-09 17:38:59.209 | INFO  | main       | verter.BmsToInverter:264   | Starting BMS receiver...
2025-02-09 17:38:59.234 | INFO  | Thread-11  | verter.BmsToInverter:290   | Reading BMS #2 SHOTO_MODBUS on /dev/ttyUSB0...
2025-02-09 17:38:59.238 | INFO  | Thread-11  | toinverter.core.Port:111   | Opening /dev/ttyUSB0 ...
2025-02-09 17:38:59.308 | INFO  | Thread-11  | toinverter.core.Port:113   | Opening port /dev/ttyUSB0 SUCCESSFUL
2025-02-09 17:38:59.426 | ERROR | Thread-11  | toBmsModbusProcessor:46    | Error reading from modbus!
java.io.IOException: Error reading from modbus device #2(address: 30001, numRegisters: 1)
        at com.airepublic.bmstoinverter.protocol.modbus.J2ModMasterPort.readHoldingRegisters(J2ModMasterPort.java:182) ~[protocol-modbus-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.protocol.modbus.J2ModMasterPort.sendFrame(J2ModMasterPort.java:124) ~[protocol-modbus-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.bms.shoto.modbus.ShotoBmsModbusProcessor.sendMessage(ShotoBmsModbusProcessor.java:52) ~[bms-shoto-modbus-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.bms.shoto.modbus.ShotoBmsModbusProcessor.collectData(ShotoBmsModbusProcessor.java:38) ~[bms-shoto-modbus-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$3(BmsToInverter.java:291) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]
2025-02-09 17:38:59.445 | INFO  | main       | verter.BmsToInverter:312   | Starting inverter sender...
2025-02-09 17:38:59.445 | INFO  | Thread-11  | verter.BmsToInverter:342   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff

2025-02-09 17:38:59.446 | INFO  | Thread-11  | verter.BmsToInverter:412   | BMS alarms:
        NONE
```

Wonder if it's the "30000 +"?  Does it want to be '40000 +'?

https://github.com/ai-republic/bms-to-inverter/blob/main/bms-shoto-modbus/src/main/java/com/airepublic/bmstoinverter/bms/shoto/modbus/ShotoBmsModbusProcessor.java#L38C11-L45C40

If I get some dedicated time later I'll see about playing with it.

hpeyerl, 2025-02-10:
I used the modbus-cli tool to query the serial number from the battery at Address 2:

```
# modbus -s 2 -b 9600 -p 1 -P n  /dev/ttyUSB0 -v h@4197/10H
Serial port /dev/ttyUSB0. Parameters: 9600 baud, 1 stop bit(s), parity: N, timeout 5.0s.
_ < 02 03 10 65 00 0a d1 21 >
_ < 02 03 14 36 39 34 34 34 32 36 33 30 34 31 30 33 31 30 31 34 34 30 5e 0a 05 > 25 bytes
_ [13881, 13364, 13362, 13875, 12340, 12592, 13105, 12337, 13364, 12382]
4197: (13881, 13364, 13362, 13875, 12340, 12592, 13105, 12337, 13364, 12382)
```

4197 == 0x1065

So it apears that we don't need to offset the register address by 30000 or 40000.  (based on "02 03 10 65")

I tried to figure out how to recompile the Shoto-modbus jar but I don't know how to use Maven to do that.

ai-republic, 2025-02-15:
@hpeyerl I changed the code so that the register start range for the holding register starts at 0 not 30000. Could you try that? Just do a _Clean Install_ on the _Configurator_ app.

hpeyerl, 2025-02-15:
Ok.  It's actually sending queries to the BMS at addr=2 and receiving responses.  It doesn't seem to be parsing the responses correctly.

```
2025-02-15 13:31:16.957 | INFO  | Thread-11  | toinverter.core.Port:111   | Opening /dev/ttyUSB0 ...
2025-02-15 13:31:17.103 | INFO  | Thread-11  | toinverter.core.Port:113   | Opening port /dev/ttyUSB0 SUCCESSFUL
2025-02-15 13:31:17.897 | INFO  | main       | verter.BmsToInverter:312   | Starting inverter sender...
2025-02-15 13:31:17.909 | INFO  | Thread-14  | verter.BmsToInverter:316   | Sending to inverter NONE on can0...
2025-02-15 13:31:18.015 | INFO  | Thread-11  | verter.BmsToInverter:342   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       100.0  0.0     0.0     0.0(#0)         0.0(#0) 0.0
```

When I read register 0x1 from the battery using modbus-cli:

```
_ < 02 03 00 01 00 01 d5 f9 >
_ < 02 03 02 15 3b b3 07 > 7 bytes
_ [5435]
1: 5435 0x153b
```

When I run strace on java to find the read/writes to the serial port, I see:

```16498 write(1, "2025-02-15 13:31:17.103 | INFO  | Thread-11  | toinverter.core.Port:113   | Opening port /dev/ttyUS"..., 113) = 113
16498 write(33, "2025-02-15 13:31:17.103 | INFO  | Thread-11  | toinverter.core.Port:113   | Opening port /dev/ttyUS"..., 113) = 113
16498 write(49, "\x02\x03\x00\x01\x00\x01\xd5\xf9", 8) = 8
16498 read(49, "\x02", 1)               = 1
16498 read(49, "\x03", 1)               = 1
16498 read(49, "\x02", 1)               = 1
16498 read(49, "\x15\x3a\x72\xc7", 4)   = 4
```

You can see that the app read:

<02 03 02 15 3a 72 c7>

(Vbat changed by .01v between when bms-to-inverter read it and when modbus-cli read it).

So bms-to-inverter received 0x153a which is 54.34v

[image]

ai-republic, 2025-02-16:
Could you please attach the log so that I can analyse what coming in and how its processed?

hpeyerl, 2025-02-16:
[BMS-to-Inverter.log](https://github.com/user-attachments/files/18814675/BMS-to-Inverter.log)

ai-republic, 2025-02-24:
Thanks @hpeyerl,
I think I figured it out now. Could you please do _Clean install_ and test again?

hpeyerl, 2025-02-24:
Seemingly no change.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/18946145/BMS-to-Inverter.log)

```
2025-02-24 15:49:07.482 | INFO  | Thread-11  | verter.BmsToInverter:290   | Reading BMS #2 SHOTO_MODBUS on /dev/ttyUSB0...
2025-02-24 15:49:07.486 | INFO  | Thread-11  | toinverter.core.Port:111   | Opening /dev/ttyUSB0 ...
2025-02-24 15:49:07.555 | INFO  | Thread-11  | toinverter.core.Port:113   | Opening port /dev/ttyUSB0 SUCCESSFUL
2025-02-24 15:49:07.625 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 00 01 00 01 D5 F9
2025-02-24 15:49:07.660 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 02 14 A2 72 FD
2025-02-24 15:49:07.721 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 00 05 00 02 D4 39
2025-02-24 15:49:07.756 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 04 00 0F 00 0E 78 F4
2025-02-24 15:49:07.809 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 00 12 00 20 E4 24
2025-02-24 15:49:07.900 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 40 00 0F 00 0F 00 0E 00 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0C E3 0C E1 0C E2 0C E3 0C E1 0C E3 0C E4 0C E2 0C E3 0C E4 0C E4 0C E4 0C E3 0C E4 0C E5 0C E3 7F 2D
2025-02-24 15:49:07.954 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 01 01 00 02 94 04
2025-02-24 15:49:07.979 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 04 33 43 00 11 F7 AF
2025-02-24 15:49:08.032 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 01 0F 00 01 B5 C6
2025-02-24 15:49:08.059 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 02 00 10 FD 88
2025-02-24 15:49:08.113 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 10 10 00 01 81 3C
2025-02-24 15:49:08.139 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 02 14 B4 F3 33
2025-02-24 15:49:08.194 | DEBUG | Thread-11  | o.ModbusRTUTransport:266   | Sent: 02 03 10 31 00 12 90 FB
2025-02-24 15:49:08.316 | DEBUG | Thread-11  | o.ModbusRTUTransport:443   | Response: 02 03 24 FD 63 27 10 00 40 1F DC 27 10 00 04 00 00 00 02 00 00 00 00 00 00 00 00 02 03 FF FF 00 08 00 00 17 89 00 00 F7 AD
2025-02-24 15:49:08.326 | INFO  | main       | verter.BmsToInverter:312   | Starting inverter sender...
2025-02-24 15:49:08.359 | INFO  | Thread-14  | verter.BmsToInverter:316   | Sending to inverter SOLARK_CAN on can0...
2025-02-24 15:49:08.386 | DEBUG | Thread-14  | verter.core.Inverter:193   | Inverter SOLARK_CAN send: Buffer (HEX): [0x51, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-24 15:49:08.388 | INFO  | Thread-14  | toinverter.core.Port:111   | Opening can0 ...
2025-02-24 15:49:08.439 | INFO  | Thread-11  | verter.BmsToInverter:342   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 100.0	0.0	0.0	0.0(#0) 	0.0(#0)	0.0 
```

I checked that the Shoto .jar is the same as the one in the compile output directory so I know the build completed and the files got transferred across to the Pi correctly.

ai-republic, 2025-03-03:
Hmm, I added tests with exactly those responses and they show that the data is read correctly:

[image]

Could you double-check please. Use the latest _Configurator_ app to do a _Clean install_ and possibly uncheck the webserver and set your inverter to _NONE_ for now.
Thanks @hpeyerl

hpeyerl, 2025-03-09:
Sorry for the delay in doing this.  I've had a perfect storm of things that needed to get done ahead of this.  I just tried it again.  I removed the directory from my Pi, and I removed everything from my build machine and downloaded a fresh configurator.jar (after saving/restoring my config.properties).  I did a build and rsync'd the files to my Pi.  No change.

Here's the log.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/19150279/BMS-to-Inverter.log)

hpeyerl, 2025-03-22:
If I had some way to build this from source, I could probably have this fixed in no time at all.  I can't seem to get Maven to want to build this.  If we somehow had the option to change this to point at a local directory:

https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/src/main/java/com/airepublic/bmstoinverter/configurator/Configurator.java#L257

...then that would probably help.  Not sure though.

ai-republic, 2025-03-31:
@hpeyerl I've modified quite a lot last weekend and also pushed a new _Configurator_ app. Could you please download it [here](https://github.com/ai-republic/bms-to-inverter/tree/main/configurator/current)

hpeyerl, 2025-04-02:
I erased everything and started from scratch twice just to be double sure using md5sum along the way to ensure that the configurator.jar that I was running was the one downloaded from github, and that the library files the configurator built and that I subsequently copied onto my pi, were all the same.

No change in behavior.  Looking at the log, it does appear that the responses coming from the battery pack are correct; and they just aren't being parsed correctly.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/19571340/BMS-to-Inverter.log)

ai-republic, 2025-04-09:
Hi @hpeyerl,
I think I found the issue. The tests were working fine, but that was with the battery pack 0. I've now fixed the code to use the unitId.
Could you please do a _Clean install_ and see what comes out?

hpeyerl, 2025-04-09:
Still no change.  Same process.  Delete everything, download fresh configurator.jar, copy my config.properties in, config/build/clean-install, rsync, start.sh.

```
2025-04-09 16:51:50.047 | INFO  | Thread-6   | verter.BmsToInverter:333   | Sending to inverter NONE on can0...
2025-04-09 16:51:50.762 | INFO  | Thread-3   | verter.BmsToInverter:307   | Reading BMS #2 SHOTO_MODBUS on /dev/ttyUSB0...
2025-04-09 16:51:51.051 | INFO  | Thread-6   | verter.BmsToInverter:333   | Sending to inverter NONE on can0...
2025-04-09 16:51:51.496 | INFO  | Thread-3   | verter.BmsToInverter:359   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 100.0	0.0	0.0	0.0(#0) 	0.0(#0)	0.0 

2025-04-09 16:51:51.500 | INFO  | Thread-3   | verter.BmsToInverter:427   | BMS alarms:
	BMS #1:	ALARM -> CELL_VOLTAGE_HIGH
	BMS #1:	ALARM -> PACK_VOLTAGE_HIGH
	BMS #1:	WARNING -> CHARGE_TEMPERATURE_LOW
	BMS #1:	WARNING -> DISCHARGE_TEMPERATURE_HIGH
	BMS #1:	WARNING -> PACK_TEMPERATURE_HIGH
	BMS #1:	ALARM -> CHARGE_CURRENT_HIGH
	BMS #1:	WARNING -> CHARGE_TEMPERATURE_HIGH
	BMS #1:	ALARM -> SOC_HIGH
	BMS #1:	ALARM -> FAILURE_SENSOR_DISCHARGE_MODULE_TEMPERATURE
	BMS #1:	ALARM -> DISCHARGE_CURRENT_HIGH
	BMS #1:	ALARM -> PACK_VOLTAGE_LOW
	BMS #1:	ALARM -> CELL_VOLTAGE_LOW
	BMS #1:	ALARM -> FAILURE_OTHER

```

hpeyerl, 2025-04-11:
Ok.  I've figured out enough to get it to compile so if I change ShotoBmsModbusProcessor.java I can get a new .jar and scp it to my Pi.

It was a simple change to get readBatteryVoltage() to work:

```
-        pack.packVoltage = frame.getChar() / 10;
+        pack.packVoltage = frame.getInt() / 10 ;
```

Now I get:

```
2025-04-11 14:07:32.540 | INFO  | Thread-3   | verter.BmsToInverter:359   |
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 100.0	52.3	0.0	0.0(#0) 	0.0(#0)	0.0
```

I'm out of time for today.   Maybe that gets you closer to understanding what's going on.

hpeyerl, 2025-04-19:
I've done a bit more experimentation on my fork https://github.com/hpeyerl/bms-to-inverter/commit/97a2ca4a498b41aa22d0e9251712a896ba2014c8   Please excuse my debug LOG.info()'s.

Since toBuffer  (https://github.com/ai-republic/bms-to-inverter/blob/main/protocol-modbus/src/main/java/com/airepublic/bmstoinverter/protocol/modbus/ModbusUtil.java#L83) already stores everything as ints, I think we need to use getInt() instead of get{Char,Short} and then convert from Int to Char/Short (which I've been doing by casting which seems wrong but I don't speak Java).

Anyway, debug output looks better but CellMinV, CellMaxV, CellDiff, moduleRatedCapacityAh, and Alarms still aren't working correctly.

```
2025-04-17 14:28:51.813 | INFO  | Thread-6   | verter.BmsToInverter:333   | Sending to inverter NONE on can0...
2025-04-17 14:28:51.891 | INFO  | Thread-3   | verter.BmsToInverter:307   | Reading BMS #2 SHOTO_MODBUS on /dev/ttyUSB0...
2025-04-17 14:28:51.945 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 00 01 00 01 D5 F9
2025-04-17 14:28:51.970 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 02 14 AD 32 F9
2025-04-17 14:28:52.023 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 00 05 00 02 D4 39
2025-04-17 14:28:52.050 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 04 00 11 00 0F D9 32
2025-04-17 14:28:52.103 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 00 12 00 20 E4 24
2025-04-17 14:28:52.194 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 40 00 10 00 11 00 10 00 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0C E7 0C E8 0C E9 0C EA 0C EA 0C EA 0C E9 0C E9 0C EA 0C E9 0C EA 0C E9 0C EA 0C ED 0C EC 0C E9 07 27
2025-04-17 14:28:52.249 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 01 01 00 02 94 04
2025-04-17 14:28:52.289 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 04 33 43 00 11 F7 AF
2025-04-17 14:28:52.343 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 01 0F 00 01 B5 C6
2025-04-17 14:28:52.369 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 02 00 10 FD 88
2025-04-17 14:28:52.423 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 10 10 00 01 81 3C
2025-04-17 14:28:52.449 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 02 14 B4 F3 33
2025-04-17 14:28:52.502 | DEBUG | Thread-3   | o.ModbusRTUTransport:266   | Sent: 02 03 10 31 00 12 90 FB
2025-04-17 14:28:52.561 | DEBUG | Thread-3   | o.ModbusRTUTransport:443   | Response: 02 03 24 FE 88 27 10 00 4E 26 2A 27 10 00 04 00 00 00 02 00 00 00 00 00 00 00 00 02 03 FF FF 00 08 00 00 1C 27 00 00 E1 70
2025-04-17 14:28:52.565 | INFO  | Thread-3   | toBmsModbusProcessor:150   | Alarm1: 0

2025-04-17 14:28:52.567 | INFO  | Thread-3   | toBmsModbusProcessor:166   | Alarm2: 2

2025-04-17 14:28:52.569 | INFO  | Thread-3   | toBmsModbusProcessor:184   | Alarm3: 0

2025-04-17 14:28:52.572 | INFO  | Thread-3   | toBmsModbusProcessor:211   | Alarm4: 0

2025-04-17 14:28:52.575 | INFO  | Thread-3   | toBmsModbusProcessor:213   | Alarm5: 0

2025-04-17 14:28:52.577 | INFO  | Thread-3   | toBmsModbusProcessor:234   | protection:
2025-04-17 14:28:52.580 | INFO  | Thread-3   | toBmsModbusProcessor:250   | moduleRatedCapacityAh:  65535
2025-04-17 14:28:52.582 | INFO  | Thread-3   | verter.BmsToInverter:359   |
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 97.7	52.9	-3.7	0.0(#0) 	0.0(#0)	0.0

2025-04-17 14:28:52.587 | INFO  | Thread-3   | verter.BmsToInverter:427   | BMS alarms:
	BMS #1:	ALARM -> PACK_VOLTAGE_LOW
	BMS #1:	ALARM -> CELL_VOLTAGE_HIGH
	BMS #1:	ALARM -> FAILURE_SENSOR_DISCHARGE_MODULE_TEMPERATURE
	BMS #1:	WARNING -> PACK_TEMPERATURE_HIGH
	BMS #1:	WARNING -> DISCHARGE_TEMPERATURE_HIGH
	BMS #1:	ALARM -> PACK_VOLTAGE_HIGH
	BMS #1:	WARNING -> CHARGE_TEMPERATURE_HIGH
	BMS #1:	ALARM -> FAILURE_OTHER
	BMS #1:	ALARM -> CELL_VOLTAGE_LOW
	BMS #1:	ALARM -> DISCHARGE_CURRENT_HIGH
	BMS #1:	WARNING -> CHARGE_TEMPERATURE_LOW
	BMS #1:	ALARM -> SOC_HIGH
	BMS #1:	ALARM -> CHARGE_CURRENT_HIGH
```

ai-republic, 2025-04-20:
I followed your suggestion and converted all reads to ints and cast down to the documented data type.
Maybe you can update and test the latest changes?
Thanks 😄

hpeyerl, 2025-04-28:
sorry for the delay.

Here's the log from a fresh build/install using the current configurator.jar.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/19939497/BMS-to-Inverter.log)

I was not running on battery at the time which is probably why "A"=0

As you can see, results are basically the same as with my fork.  It's incorrectly parsing the CellMinV, CellMaxV, and alarms..

ai-republic, 2025-05-02:
ok, well looks a little better 😉 
The ModBus bytes look good also. I think I have an idea where the problem is.
Could you please update again - I added some extra logs to make it easier to determine where the error lies.
Please attach the new log.

hpeyerl, 2025-05-04:
[BMS-to-Inverter.log](https://github.com/user-attachments/files/20028201/BMS-to-Inverter.log)

ai-republic, 2025-05-12:
Ah, sorry, I only had some System.outs that are not written to the log file. I've updated it to log to the log file now so I would need you to update again and send another log - sorry about that

hpeyerl, 2025-05-14:
[BMS-to-Inverter.log](https://github.com/user-attachments/files/20209979/BMS-to-Inverter.log)

ai-republic, 2025-05-17:
Alright thanks a lot. This really helps now to understand the problem!
I'll try to fix that next weekend as I have no time this weekend

hpeyerl, 2025-06-01:
Just checking in if you have anything new to test on this?