ai-republic/bms-to-inverter #98: Growatt Mod 10KTL-XH setup

romanbetwo, 2025-02-04:
Hi again finaly I get hands on my own MOD puppy, I would love to take a shot with simulate battery and test implemented growatt modbus * HV protocol 

2025-02-04 21:26:14.681 | INFO  | Thread-3   | verter.BmsToInverter:294   | Sending to inverter GROWATT_MODBUS on /dev/ttySC0...
2025-02-04 21:26:14.684 | DEBUG | Thread-3   | verter.core.Inverter:171   | Calling inverter plugin (onBatteryAggregation): PresetBatteryPackDataPlugin
2025-02-04 21:26:14.687 | ERROR | Thread-3   | verter.core.Inverter:198   | Failed to send frame to inverter GROWATT_MODBUS :Buffer (HEX): [null]

can you fix this issue for me ? 
Thanks Torsten

cheers Roman

ps: I try  pylon_modbus -  it sends simulated data

ai-republic, 2025-02-08:
Hi @romanbetwo,
could you attach the complete log please.  I think there should be an exception after the last log entry you sent. That would tell me more about what happened. Thanks!

romanbetwo, 2025-02-13:
Hi on parallel "frontline" :)  so I try hard to catch any possible data between growatt mod and ARK HV BMS.

I know we have the "docs" but seriously I have literaly no idea where to look.

On MOD version of inverter they have dedicated RS485 port PIN 7A 8B  ----AND--- ENABLE + / -   9/10 pins  for the battery, which sounds eazy but its not. Port it self Uses 38400 speed  (confirmed by Growatt) and the data , well there is no document covering this kind of data .. 

So I try .. without connected BMS port its self runs this kind of "line"
<- 0x1 0x64 0xC 0x5D 0x0 0x64 0x0 0x0 0x80 0x64 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x29 0x4 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xCA

which gives me (but am not expert) hard time to decode

[image]

So I manage to take ARK HV BMS without battery (powered by external power supply) and turn both on.

 What I saw next "on parallel RS485 reader" is here 
<- 0x1 0x3 0x3 0x18 0x3 0x0 0x8 0x6 0x0 0x0 0x4 0x3 0x0 0x1 0x5 0x0 0x0 0x0 0x9 0x6 0x10 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1B 0x15
<- 0x1 0x3 0x10 0x3 0x0 0x0 0x0 0x11 0xB 0xD 0x9 0x13 0x6 0x6 0x6 0x6 0x6 0x6 0x12 0x6 0x13 0x18 0x6 0x6 0x6 0xD 0x14 0x0 0x0 0x1A 0xB 0x14 0x2 0xB 0x3 0x0 0x16 0x1B 0xB 0xB 0x3 0x5 0x1 0x1 0x0 0x5 0x0 0x2 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x10 0x4 0x0 0x0 0x0 0x0 0x10 0x6 0x0
<- 0x1 0x3 0x0 0xA 0x0 0x0 0x11 0x13 0xB 0x13 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x19 0x2 0xD 0xA 0x2 0x1D 0x13 0x0
<- 0x1 0x4 0x3 0x18 0x17 0x0 0x4 0x3 0x0 0x0 0x4 0x3 0x0 0x1 0x5 0x0 0x0 0x0 0x9 0x6 0x10 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xA
<- 0x1 0x4 0x8 0x0 0x1 0x2 0x2 0x1 0x15 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x3 0x1C 0x6 0x1E 0x0 0x0 0x0 0x0 0x0 0xA 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xF 0x0 0x0 0x5 0x1B 0x3 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x10 0x13 0x10 0x3 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x11 0x6 0x11 0x6 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x0 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x1 0x0 0x6 0x0 0x0 0x0 0x0 0x0 0x2 0xB 0xC 0x3 0x0 0x0 0x0 0x1B 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x3 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xA 0x17 0x5
<- 0x1 0x4 0x3 0x18 0x17 0x0 0x4 0x3 0x0 0x0 0x4 0x3 0x0 0x1 0x5 0x0 0x0 0x0 0x9 0x6 0x10 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xA
<- 0x1 0x4 0x8 0x0 0x1 0x2 0x2 0x1 0x15 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x3 0x1C 0x6 0x1E 0x0 0x0 0x0 0x0 0x0 0xA 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xF 0x0 0x0 0x5 0x1B 0x3 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x10 0x13 0x10 0x3 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x11 0x6 0x11 0x6 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x0 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x1 0x0 0x6 0x0 0x0 0x0 0x0 0x0 0x2 0xB 0xC 0x3 0x0 0x0 0x0 0x1B 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x3 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xA 0x17 0x5
<- 0x1 0x4 0x3 0x18 0x17 0x0 0x4 0x3 0x0 0x0 0x4 0x3 0x0 0x1 0x5 0x0 0x0 0x0 0x9 0x6 0x10 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xA
<- 0x1 0x4 0x8 0x0 0x1 0x2 0x2 0x1 0x15 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x3 0x1C 0x6 0x1E 0x0 0x0 0x0 0x0 0x0 0xA 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xF 0x0 0x0 0x5 0x1B 0x3 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x10 0x13 0x10 0x3 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x11 0x6 0x11 0x6 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x0 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x1 0x0 0x6 0x0 0x0 0x0 0x0 0x0 0x2 0xB 0xC 0x3 0x0 0x0 0x0 0x1B 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x3 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xA 0x17 0x5
<- 0x1 0x4 0x3 0x18 0x17 0x0 0x4 0x3 0x0 0x0 0x4 0x3 0x0 0x1 0x5 0x0 0x0 0x0 0x9 0x6 0x10 0x0 0x0 0x1 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xA
<- 0x1 0x4 0x8 0x0 0x1 0x2 0x2 0x1 0x15 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x3 0x1C 0x6 0x1E 0x0 0x0 0x0 0x0 0x0 0xA 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xF 0x0 0x0 0x5 0x1B 0x3 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x10 0x13 0x10 0x3 0x0 0x0
<- 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x11 0x6 0x11 0x6 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x0 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x1 0x0 0x6 0x0 0x0 0x0 0x0 0x0 0x2 0xB 0xC 0x10 0x0 0x0 0x0 0x1B 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0x3 0x0 0x0 0x0 0x0 0x0 0x0
<- 0x0 0x10 0x6 0x1D 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1E 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x8 0x0 0x0 0x0 0x0 0x0 0 [... cut ...]
I am sure that more of data will be available if I use real battery because off cell informations are  missing //voltage temps and stuff, but I just wanted to see whats going on and try to understand. At this moment its hard for me tu get testing ARK HV battery so thats why I "just" try with PSU.
Inverter however pop up Storage system with fault but it reads the voltage and some SOC%
I did not connect BMS terminals to inverter, just .. I dont want any fireworks to be clear.

I will update previouse post after Pi reinstall :)

romanbetwo, 2025-04-14:
Just a little update, I am still fighting Growatt to give me HV protocol for their ark pack (MOD, MIN) 
There has been some infrastructure changes so its hard to gain someone trust but I am trying my best.
Just to know the HV protocol runs at 38400 speed if someone wants to decode :) 

I will inform asap when I get more info.

Regards team 🫡✌️

closed 2026-01-30 by ai-republic