ai-republic/bms-to-inverter #99: Support for SolarEdge SE10K-RWB48BFN4 and JK-BMS PB1A16S-10P.

Grzesio1989, 2025-02-05:
Hello, I need your help with connecting SolarEdge SE10k-RWB48BFN4 with a battery equipped with jk-bms PB1A16S-10P.

 I am attaching the original can protocol file and translated tables of the registers. 
[1559288117.pdf](https://github.com/user-attachments/files/18673016/1559288117.pdf)
[jikong_BMS_CanV2.0.pdf(https://github.com/userattachments/files/18673009/jikong_BMS_CanV2.0.pdf).

The inverter, according to the manual, supports CAN communication with LG CHEM RESU and BYD Battery-box Premium LV.

 I found only this thread about the BYD can protocol:

https://github.com/dfch/BydCanProtocol/blob/main/README.md

I look forward for Your support:)

Thank You!

ai-republic, 2025-02-08:
Hi @Grzesio1989,
have you tried the _JK_CAN_ BMS binding? That is essentially implemented. There are a couple of new commands in the spec you provided but should work fine. The _BYD_CAN_ BMS binding matches also essentially what you provided in the link. I can whip up the inverter binding from that. I hope I can get that ready for you tomorrow 😄

Grzesio1989, 2025-02-08:
Hi!
Yes, i tried but it returns wrong value:

2025-02-08 19:09:00.515 | INFO  | Thread-11  | verter.BmsToInverter:294   | Send                                                                                                                                                       ing to inverter NONE on can1...
2025-02-08 19:09:00.632 | INFO  | Thread-10  | verter.BmsToInverter:268   | Read                                                                                                                                                             ing BMS #1 JK_CAN on can0...
2025-02-08 19:09:00.637 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       -20.0  204.8   -400.0  0.21(#12)       2.048(#0)       1.838

2025-02-08 19:09:00.639 | INFO  | Thread-10  | verter.BmsToInverter:390   | BMS                                                                                                                                                              alarms:
        NONE

I also tried PYLON_CAN binding (changed on bms too) and it was better but still show wrong alarms:

2025-02-08 19:24:54.331 | INFO  | Thread-11  | verter.BmsToInverter:294   | Sending to inverter NONE on can1...
2025-02-08 19:24:54.444 | INFO  | Thread-10  | verter.BmsToInverter:268   | Reading BMS #1 PYLON_CAN on can0...
2025-02-08 19:24:54.448 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       51.0   49.2    0.0     3.282(#4)       3.282(#2)       0.0

2025-02-08 19:24:54.451 | INFO  | Thread-10  | verter.BmsToInverter:388   | BMS alarms:
        BMS #1: ALARM -> CELL_VOLTAGE_LOW
        BMS #1: ALARM -> CELL_TEMPERATURE_LOW
        BMS #1: ALARM -> CHARGE_CURRENT_HIGH
        BMS #1: ALARM -> CELL_VOLTAGE_HIGH
        BMS #1: ALARM -> DISCHARGE_CURRENT_HIGH

ai-republic, 2025-02-09:
Hi @Grzesio1989,
I fixed some stuff and added v2.1 support to the _JK_CAN_ BMS binding. 
I also added the _BYD_CAN_ inverter binding. Please do a _Clean install_ on the updated _Configurator_ app but only using the JK BMS binding and NONE for the inverter for now to see if we get good values now

Grzesio1989, 2025-02-09:
Hi @ai-republic !
I made Clean install but unfortunately, the readings are still wrong.
The log file:  [BMS-to-Inverter.log](https://github.com/user-attachments/files/18724168/BMS-to-Inverter.log)

I also found that bytes shold be swaped:

`2025-02-09 11:45:30.805 | INFO  | Thread-12  | verter.BmsToInverter:316   | Sending to inverter NONE on can1...
2025-02-09 11:45:30.872 | INFO  | Thread-11  | verter.BmsToInverter:290   | Reading BMS #1 JK_CAN on can0...
2025-02-09 11:45:30.874 | DEBUG | Thread-11  | ocol.can.JavaCANPort:76    | CAN frame read...
2025-02-09 11:45:30.877 | DEBUG | Thread-11  | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0xF4, 0x04, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xD1, 0x0C, 0x01, 0xD2, 0x0C, 0x04, 0x00, 0x00]

2025-02-09 11:45:30.879 | INFO  | Thread-11  | verter.BmsToInverter:342   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 -20.0	204.8	400.0	0.209(#12) 	2.048(#0)	1.839 

                                                                   BYTE:   1        2        3       4        5       6        7       8
                                                                      BIT:  0-7   8-15 16-23 24-31 32-39
[0xF4, 0x04, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xD1, 0x0C, 0x01, 0xD2, 0x0C, 0x04, 0x00, 0x00]

ID: 0x04F4

Max cell Volt:
Hex: 0x0C,0xD1 
Dec: 3281 mV

MaxCvNO:
Hex: 0x01
Dec: 1

MinCellVolt:
Hex: 0x0C, 0xD1
Dec: 3281 mV

MinCvNO:
Hex: 0x04
Dec: 4

2025-02-09 11:45:29.802 | INFO  | Thread-12  | verter.BmsToInverter:316   | Sending to inverter NONE on can1...
2025-02-09 11:45:29.866 | INFO  | Thread-11  | verter.BmsToInverter:290   | Reading BMS #1 JK_CAN on can0...
2025-02-09 11:45:29.867 | DEBUG | Thread-11  | ocol.can.JavaCANPort:76    | CAN frame read...
2025-02-09 11:45:29.869 | DEBUG | Thread-11  | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0xF4, 0x02, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xEC, 0x01, 0xA0, 0x0F, 0x33, 0x00, 0x00, 0x00]

2025-02-09 11:45:29.870 | INFO  | Thread-11  | verter.BmsToInverter:342   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 -20.0	204.8	400.0	0.209(#12) 	2.048(#0)	1.839 

                                                                     BYTE:  1      2        3       4       5       6       7       8
                                                                     BIT:  0-7  8-15  16-23 24-31 32-39
[0xF4, 0x02, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xEC, 0x01, 0xA0, 0x0F, 0x33, 0x00, 0x00, 0x00]

ID 0x02F4

BattVolt:
Hex: 0x01, 0xEC,
Dec: 492 (Res 0.1) = 4.92V

BatCurr:
Hex: 0x0F, 0xA0
Dec: 400 (offset -400) = 0A

SOC:
Hex: 0x33
Dec: 51`

Maybe this will help

ai-republic, 2025-02-15:
Hi @Grzesio1989 ,
I changed the byte ordering to little endian. Now you should get correct values :)

Grzesio1989, 2025-02-15:
Hi @ai-republic ,
Unfortunetly problem still exist. I made cleen install and i get identical values.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/18809468/BMS-to-Inverter.log)

ai-republic, 2025-02-16:
Ah sorry, I put the byte ordering on the frame but not on the copied data payload. My bad 😉 
Can you run a _Clean install_ again and test please?

Grzesio1989, 2025-02-16:
I tried and something is still wrong. 

[BMS-to-Inverter.log](https://github.com/user-attachments/files/18814172/BMS-to-Inverter.log)

ai-republic, 2025-02-17:
@Grzesio1989 thanks for the Log! I fixed and tested it. Sorry I missed to read the skip bytes 😉 
Now your data will be correct

Grzesio1989, 2025-02-17:
Hi @ai-republic, 
Great job!
Now i can see correct data:) But sometimes the current values are too big :( 

[BMS-to-Inverter.log](https://github.com/user-attachments/files/18827551/BMS-to-Inverter.log)

Is it possible to read individual cell data?  

In few next weeks i will try to connect it to inventer and i'll give You feedback how it's works:) But firstly i have to make propoer place for it

ai-republic, 2025-02-24:
Not sure why the current looks funny. The spec says this and I tested it with these and it give the correct value (56.7A):

[image]

Ronald-3D, 2026-09-27:
Altough closed, hope to get a respons.
I have the same configuration and i hope someone can tell me how configure and wire up the set.
I think LG and BYD are no longer supported on the CAN bus. instead you have have to use the rs485.

Who can help me on my way?

Thanls.