ai-republic/bms-to-inverter #102: Support for MPP Solar Lib protocol

carsonfee, 2025-02-16:
Looking for Lib support for MPP Solar. The MPP inverter I have is an Lvx6048WP their docs say thepis model supports Pylon but I have tried that and it doesn't seem to be reading the SOC correctly so it doesn't ever initiate a charge cycle

carsonfee, 2025-02-16:
[BMS communication protocol 20201202.pdf](https://github.com/user-attachments/files/18814138/BMS.communication.protocol.20201202.pdf)

This looks to be the protocol spec

ai-republic, 2025-02-17:
Welcome back @carsonfee 😄 
yes. this should be easily doable as is a RS485/Modbus protocol.
Just give me some time as I'm quite busy currently. Maybe I can get a start on the weekend

carsonfee, 2025-02-17:
Sweet. Got rid of the skybox which never worked. They actually never truly
supported RS485.  The MPP inverter has been pretty good so far, just need
to get it to read the correct SOC.

ai-republic, 2025-03-03:
@carsonfee I've implemented an initial version that should give you the cell voltages, temperatures, SOC, current and voltage.
Please feel free to test it by downloading the latest _Configurator_ app and do a _Clean install_. Let me know how it goes 😉

carsonfee, 2025-03-03:
Thanks, I'll check it out!

ai-republic, 2025-03-09:
@carsonfee finished the full _MPP_SOLAR_ binding implementation now.

carsonfee, 2025-03-09:
Just put a fresh OS on my Pi4B+ I downloaded current configurator and did a clean install.  I don't see the MPP-Modbus in the inverter list.

ai-republic, 2025-03-09:
I see it:

[image]

Have you downloaded from here: https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/current/configurator.jar ?

carsonfee, 2025-03-09:
> I see it:
>
> image.png (view on web)
>
> Have you downloaded from here:
> https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/current/configurator.jar
> ?

yes

[image: image.png]

carsonfee, 2025-03-09:
[image]

Don't see it in either BMS or Inverter

carsonfee, 2025-03-09:
I set my batteries to pylon on both can and rs485.  RS485 didn't seem to read but can does. Here's a dump

  can0  355   [8]  5A 00 64 00 00 00 00 00
  can0  356   [8]  AF 14 A8 FF A8 00 00 00
  can0  35C   [8]  C0 00 00 00 00 00 00 00
  can0  35E   [8]  4C 45 4F 43 48 20 20 20
  can0  359   [8]  00 00 00 00 04 50 4E 00
  can0  351   [8]  34 02 10 04 A0 0F B0 01
  can0  355   [8]  5A 00 64 00 00 00 00 00
  can0  356   [8]  AF 14 A8 FF A8 00 00 00
  can0  35C   [8]  C0 00 00 00 00 00 00 00
  can0  35E   [8]  4C 45 4F 43 48 20 20 20
  can0  359   [8]  00 00 00 00 04 50 4E 00
  can0  351   [8]  34 02 10 04 A0 0F B0 01
  can0  355   [8]  5A 00 64 00 00 00 00 00

output seems to be parsed wrong.

[image]

I am connected to the inverter on rs485 and I set it to expect pylon but I'm not sure if that's reading right

here's the log.

[BMS.log](https://github.com/user-attachments/files/19153005/BMS.log)

I tried running the configurator on my PC as well as the PI and neither one showed the MPP solar

carsonfee, 2025-03-10:
batteries a using pylon on rs485.  These are Sungold SG48100 rack batteries. 

dumping ttySC0 shows a mess, not sure if that's incoming or outgoing or both

cat /dev/ttySC0
000000000000000000000000000000000000ï¿½ï¿½ï¿½ï¿½29ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½Ýµï¿½0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000~20004600824234383130302D393533385368656E5A68656E20504143452020202020202000091000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000<ï¿½ï¿½wvï¿½ï¿½ï¿½ï¿½ßswï¿½ï¿½ï¿½66ï¿½ï¿½~20004600824234383130302D393533385368656E5A68656E2050414345202020202020200009100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003ï¿½ï¿½ï¿½ï¿½ï¿½6ï¿½ßï¿½ï¿½ï¿½ï¿½ï¿½3f0000000000000000000000000000000000000000000000000000000000000000000000000~2000460082423438313030285368656E20504143452020202020202000091000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003ï¿½ï¿½ï¿½ï¿½ï¿½6ï¿½ßï¿½ï¿½ï¿½ï¿½ï¿½3Ì0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

Sungold has a confusing note on RS485.

7.3.2.RS485 communication 
With dual RS485 interface, you can view the information of PACK, the default baud rate is 9600bps. 
To communicate with the monitoring device through RS485, the monitoring device is used as the host 
to poll data according to the address, and the address setting range is 2~63.

I have no idea if they mean the address is the ID set on the dip switches or some other thing. going to email their tech support.

ai-republic, 2025-03-23:
Hi @carsonfee, just a question: how many battery packs do you have? 2 or 4? And how many did you configure and what is the configured ID?

carsonfee, 2025-03-23:
4 packs. ID 1,2,3,4. 1 is the master. I just got a new bunch of cables to
put all of them on the can bus. They are connected to each other by rs485
but I don't think the master is sending the status of the slaves from the
can

ai-republic, 2025-03-31:
@carsonfee did you get a chance to test the MPPSOLAR_MODBUS binding?

carsonfee, 2025-03-31:
Not yet, I just got a firmware update for the inverter that's supposed to
fix up some of their protocol support