ai-republic/bms-to-inverter #106: Growatt ARK and AXE Clusters

77pippofranco, 2025-03-04:
Excuseme if i write here

I have 3 parallel spf5000es with hub lv a1

2 clusters of growatt ARK
5 x growatt ark 2,5L

1 cluster of growatt AXE
2 x growatt axe 5.0 L

Axe can’t connect to Hub 

Problem regarding A charge

Sometime get 200A ( 66A each inverter )

Sometime get 120A ( 40A each nverter )

Can i use bms-to-inverter for set inverter to charge 300A ( 100A each inverter ) ?

ai-republic, 2025-03-06:
Yes, you could use the application and aggregate the AXE and use the `ManipulateBatteryPackDataPlugin` to manipulate what's going out to the inverter and set the _Max. charge current_ to 300A.
That all depends on if your BMSes have compatible protocols.

77pippofranco, 2025-03-06:
Tested in debug but not solve my problem. It working in the midle betwen battery and inverter. 

Connected betwen hub / battery do not working

Maybe also my old rpi B+ v1.2 armv6l 😔

Project is great 👍. Thx for support

ai-republic, 2025-03-09:
I would suggest a setup like this:

[image]

My recommendation would be a PI4B+ with 2x 2CH-CAH-FD hat so that you have up to 4 CAN ports. This way you can connect the ARK battery clusters/Hub LV A1 and the AXE battery cluster (can1) directly to the PI and from the PI on can2 to your SPF inverter (master)