ai-republic/bms-to-inverter #107: Support for Discover Energy Systems LifePO4

ryanpulley, 2025-03-06:
Hi @ai-republic 

You have developed some phenomenal software here.   Great work!

I have recently refreshed my system to include a [Midnite Solar MN15-12KW-AIO All in One](https://www.midnitesolar.com/productPhoto.php?product_ID=766&productCat_ID=65&sortOrder=1) inverter.   My old inverter was not capable of communication with a BMS.

I have 3 [Discover Energy System AES rack mount batteries](https://discoverenergysys.com/products/lithium-batteries/aes-rackmount) for a total of 300ah.    These batteries communicate with a [Discover Lynx II Gateway](https://discoverenergysys.com/products/communication-solutions/lynk-2-gateway-solar) which handles all communication over CAN / RS485 with various inverters.   Most of those communication protocols are not documented and have been co-developed with various inverter / charge controller vendors.   However, they do publish a protocol for others to integrate with which they call "Discover CAN Open":

[800-0010 (LYNK CANopen Interface Technical Reference Manual) - Rev D.pdf](https://github.com/user-attachments/files/19112290/800-0010.LYNK.CANopen.Interface.Technical.Reference.Manual.-.Rev.D.pdf)

Within the Lynx II Gateway, I'm able to modify the protocol that is being used and some attributes of those protocols.

The Midnite Solar MN15-12KW-AIO inverter can communicate using the PylonTech protocol.   I made an attempt to select a protocol just called "P-Generic" from the Lynx II gateway and connect the BMS cable to the inverter with limited success.   It's communicating SOC, but I'm getting very strange behaviors when charging the batteries from the inverter.    It takes a steep dive on charge amps being supplied in the range of 25-70%.   At 70%, the inverter is only supplying 300W to the battery, so it would take hours to complete a charge (and suck a lot of gas from the generator when solar is not available).   I'm hoping you can help me remedy this through some support for the Discover CANOpen spec.

I do have the bms-to-inverter software installed on an RPI-4 with the Waveshare 2 channel CAN adapter you suggested and have some communication working with the battery using the P-Generic protocol on the Lynx2.   Viewing via the BMS-to-inverter web server I get SOC, Pack Voltage, and Pack Current.   Max Cell Voltage, Cell Dif and the individual Cell Voltages (3 displayed) are always 0.   Charge MOS and Discharge MOS are always "OFF".   And the alarms are showing historical alarms (currently 5 of them), but not currently active alarms.   I expect support for the spec will resolve this.   I think their "P-Generic" offering is some modified and/or partially implemented form of the PylonTech protocol.

Once again, great work!

ai-republic, 2025-03-09:
@ryanpulley I will need to dig into the CANopen way of addressing a little as its not quite clean from the document

ryanpulley, 2025-03-10:
Hi @ai-republic 

I've done a bit more digging into some other potential options outside of Discover CANopen.   CANopen seems quite robust to get a ton of information from the battery, but it's quite complicated.

I've toggled through a variety of protocols that the Discover LYNX II device supports and performed some candump.   Their Victron protocol support seams reasonable.   Here's the candump:

`candump can0
  can0  351   [8]  2E 02 04 0B 04 0B B0 01
  can0  355   [6]  57 00 64 00 FC 21
  can0  356   [6]  46 15 27 02 D2 00
  can0  35A   [8]  A8 AA 82 02 A8 AA 82 02
  can0  35F   [4]  00 00 02 00
  can0  35E   [8]  44 49 53 43 4F 56 45 52
  can0  370   [8]  44 69 73 63 6F 76 65 72
  can0  371   [8]  20 41 45 53 00 00 00 00
`

Of note, here are the inverter/charger interfaces supported by the LYNX II device:

- Discover CANopen
- Discover Serial CAN
- ATIB Electronics
- Deye
- Generic P-LV
- Luxpower
- Morningstar
- Noark
- RV-C
- SAE J1939
- Schneider BMS
- Selectronic
- Sinoboom
- SMA
- Sol-Ark
- SPE Green Chargers
- Studer
- Sunsynk
- Victron
- Wiferion
- Xanbus

If you have another recommendation on one of these vendor specific protocols, let me know and I'll select and get a candump.

ErikssonPer, 2025-03-10:
Try the Deye protocol, atleast the high voltage version is the same as pylon which is commonly used.

ryanpulley, 2025-03-11:
Yes, I tried selecting the Deye protocol from the Lynx II device and I'm unable to get anything from candump.

ErikssonPer, 2025-03-11:
Pylon and most other protocols need a wakeup or heartbeat from the inverter side first and then the bms start to transmit data.

ryanpulley, 2025-03-14:
@ai-republic 

Per @ErikssonPer suggestion, I attempted to get the Deye protocol to wakeup.  No joy...  I noticed that it's available in their drop down list on the Lynx II device, but not in their published "compatibility" list on their website.   So, I suspect it's only partially implemented.

However, I found that some documentation on the "Serial CAN" protocol, gave it a go and got a candump.   It looks to me like the best option for a fairly straightforward implementation.

Here's the candump:

  can0  351   [8]  2E 02 04 0B 04 0B B0 01
  can0  354   [8]  2C 01 BA 00 00 00 00 00
  can0  355   [8]  3E 00 64 00 00 00 00 00
  can0  356   [8]  0A 02 11 FF BE 00 00 00
  can0  35A   [7]  AB AA FE AF AA FA FF
  can0  35E   [8]  44 49 53 43 4F 56 45 52
  can0  370   [8]  44 69 73 63 6F 76 65 72
  can0  371   [8]  20 41 45 53 00 00 00 00
  can0  372   [4]  00 00 00 02
  can0  373   [4]  01 00 00 00

To understand it a bit better, I spent a little time manually decoding some pieces of the candump sample above:

0x351 / requested  charge voltage | 558
0x351 / requested charge current | 2820
0x351 / requested max discharge current | 2820
0x351 / low battery cut out voltage | 432
0x354 / Battery Nominal Capacity | 300
0x354 / Battery Remaining Capacity | 186
0x355 / Battery State of Charge | 62
0x355 / Battery State of Health | 100
0x356 / Battery Voltage | 522
0x356 / Battery Current | -239

The spec from Discover:

[800-0008 (LYNK Serial CAN Interface Technical Reference) - Rev D (1).pdf](https://github.com/user-attachments/files/19248884/800-0008.LYNK.Serial.CAN.Interface.Technical.Reference.-.Rev.D.1.pdf)

ai-republic, 2025-03-15:
Hi @ryanpulley,
this actually conforms with the Pylon CAN binding. It doesn't have the 0x354 comand but I've added that just now so you might want to try that.

ryanpulley, 2025-03-15:
I have tried and am getting unexpected results.   If I recall correctly
voltage is showing as 5.2 instead of 52 and the cell voltages are 32.x
instead of 3.2.   I’m getting some errors as well.  I’m away from the
installation until next week.   Should I run in debug mode and provide
output?

ai-republic, 2025-03-15:
Yes, please only have the BMS binding and inverter set to NONE and no optional services. And then provide the log file please.
Can you also get a CAN dump from the AES to better understand their protocol please? Maybe that will give me better insights about their CANopen address ranges. That would be great!

ryanpulley, 2025-03-19:
Hi @ai-republic 

I was able to get a debug log and candump.   I ran these for about 1 minute.   During that seem period, I was able to monitor from the Discover Lynx device GUI to see what values should be during that period.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/19351962/BMS-to-Inverter.log)

[candump.log](https://github.com/user-attachments/files/19351972/candump.log)

A handful of things that I noticed where the Discover Serial Can messages differ from Pylontech:

* Capacity (0x354) - bytes 0-1 - Value in the log was 30 and should have been 300 (shouldn't be divided by 10)
* Remaining (0x354) bytes 2-3 - Value in the log was 30 and should have been 300 (shouldn't be divided by 10)
* Pack V (0x356) bytes 0-1 - Value in the log was 5.4 and should have been 54.4 (divide by 10 instead of 100)

* It appears as if the 0x370 Message would normally give you Min/Max Temp and Min/Max Cell Voltages with Pylontech.   With the Discover that message is for the "Battery Model Name Upper ASCII".   The Min/Max Temp and Min Max Cell Voltages aren't available on the Discover protocol.

* It appears as if the 0x371 Message would give you Max V Cell/Min V Cell with Pylontech.   With the Discover, that message is for the "Battery Model Name Lower ASCII".   There isn't a corresponding message type that would provide these values either.

ai-republic, 2025-03-23:
@ryanpulley thanks for the detailed analysis. Based on your findings I will create a separate _DISCOVER_ binding with the changes you found above.

ryanpulley, 2025-03-31:
Thanks.   I’ll do a clean install and do some tests.

ryanpulley, 2025-04-02:
Hi @ai-republic,

I've been doing some testing and encountering an exception.  This exception only occurs when I have an inverter binding as it's being thrown from the "createManufacturer" method of PylonInverterCANProcessor.java.  Also, it only occurs 
after the first successful read of the 0x35E message from the BMS which is sent every 10 seconds.

Configuration:
  BMS: DISCOVER_CAN
  Inverter: PYLON_CAN
  Plugins: None
  Services: MQTT

0x35E read in logs:

2025-04-02 13:53:32.562 | DEBUG | Thread-8   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x44, 0x49, 0x53, 0x43, 0x4F, 0x56, 0x45, 0x52]

2025-04-02 13:53:32.564 | DEBUG | Thread-8   | coverBmsCANProcessor:159   |
Manufacturer: DISCOVER

Exception: 

java.nio.BufferOverflowException: null
        at java.nio.Buffer.nextPutIndex(Buffer.java:744) ~[?:?]
        at java.nio.DirectByteBuffer.putChar(DirectByteBuffer.java:583) ~[?:?]
        at com.airepublic.bmstoinverter.inverter.pylon.can.PylonInverterCANProcessor.createManufacturer(PylonInverterCANProcessor.java:145) ~[inverter-pylon-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.pylon.can.PylonInverterCANProcessor.createSendFrames(PylonInverterCANProcessor.java:42) ~[inverter-pylon-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.Inverter.process(Inverter.java:177) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.pylon.can.PylonInverterCANProcessor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[inverter-pylon-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$5(BmsToInverter.java:334) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:1575) [?:?]

ryanpulley, 2025-04-07:
Hey @ai-republic 

I was reviewing the code in more detail and found an issue that I suspect is causing the BufferOverflowException.   It appears as if createChargeDischargeFlags is using CAN address 0x35E (Manufacturer) instead of 0x35C (Charge/Discharge Flags). 

**PylonInverterCANProcessor.java**

`    // 0x35C
    private ByteBuffer createChargeDischargeFlags(final BatteryPack pack) {
        final ByteBuffer frame = prepareSendFrame(_0x035E_);    **<--should be 0x035C**
        byte flags = 0x00;

        // request full charge
        flags = BitUtil.setBit(flags, 3, false);
        // request force charge
        flags = BitUtil.setBit(flags, 4, pack.forceCharge);
        flags = BitUtil.setBit(flags, 5, pack.forceCharge);
        flags = BitUtil.setBit(flags, 6, pack.dischargeMOSState);
        flags = BitUtil.setBit(flags, 7, pack.chargeMOSState);

        frame.put(flags);

        return frame;
    }

    // 0x35E
    private ByteBuffer createManufacturer(final BatteryPack pack) {
        final ByteBuffer frame = prepareSendFrame(0x035E);
        int idx = 0;

        while (idx < pack.manufacturerCode.length() && idx < 8) {
            frame.putChar(pack.manufacturerCode.charAt(idx));
            idx++;
        }

        return frame;
    }`

njwretnm, 2025-06-11:
@ryanpulley Did you ever get this working?  I'm planning on dusting off my Lynk2 and battery packs in the next couple weeks.

ryanpulley, 2025-06-11:
@njwretnm 

No, I'm still getting a BufferOverflowException.   I did go another route that is working well, but I would love to ultimately use this software.   Are you trying to communicate with a PylonTech protocol based inverter?

njwretnm, 2025-06-11:
I was going to connect my Lynk2 to a Schneider Xw Pro.  Obviously I can use the built in Schneider pairing in the Lynk, but i was planning on adding non-Discover batteries down the road

ai-republic, 2025-06-15:
@ryanpulley & @njwretnm I pushed the change for the dis-/charge flags to used the correct CAN id. Could you please update and let me know if its working?

closed 2026-01-30 by ai-republic