ai-republic/bms-to-inverter #108: Support for Pylon ASCII Console Mode - Rs232 - (Pylon Extra2000 & Older Pylontech LiFePO4 BMS)

ned-kelly, 2025-03-09:
Hi,

The current bms-to-inverter implementation appears to use a binary RS485 protocol with structured command frames (e.g., 0x46 0x60, 0x46 0x61) However, earlier-generation Pylontech LiFePO4 batteries, such as the Pylon Extra2000, communicate using a **plain ASCII-based serial protocol** once switched into console mode.

I have tested this with a working implementation ([this Python script](https://github.com/tomascrespo/pylon-RS232-to-mqtt/blob/main/pytes_serial.py)), which successfully retrieves data from the BMS using simple ASCII commands like pwr, bat, etc. However, bms-to-inverter does not receive valid responses and logs only timeouts when trying with my Pylon Extra2000.

## Differences in Protocols

**Current Java Implementation (Binary RS485)**
- 	Uses structured binary messages with CID1/CID2 identifiers (e.g., 0x46 0x60 for system info).
- 	Implements checksum-based validation.
- 	Expects responses in a predefined binary frame format.

**Pylon Extra2000 & Older Batteries (ASCII Console Mode)**
- 	Requires an initial command to switch into console mode before it will accept ASCII commands.
- 	After entering console mode, it expects text-based commands like pwr 1, bat 1, etc.
- 	Responses are human-readable, line-based ASCII output.

## How to Enable Console Mode on Pylontech BMS

Before communicating via ASCII, the BMS must be switched into console mode by sending a specific initialization string at 1200 baud, then switching to 115200 baud for normal communication:

```
stty -F /dev/ttyUSB0 1200 raw -echo
echo -ne "~20014682C0048520FCC3\r" > /dev/ttyUSB0  # Switch to ASCII console mode
stty -F /dev/ttyUSB0 115200 raw -echo
echo -ne 'login debug\n' > /dev/ttyUSB0           # Authenticate for debug commands
```

After this, the BMS will accept standard ASCII commands such as:

```
echo -ne 'pwr 1\n' > /dev/ttyUSB0
```
(This would  return human-readable data for power status for the first battery in the array) etc.

## Sample ASCII Commands for Pylontech BMS

Once in console mode, the following commands return useful information:

```
pylon>help
@
Local command:
bat      Battery data show - bat [pwr][index]
cmdquit  Quit the Console Mode
data     History data load - data [event/history][item]
datalist Show recorded data - datalist [event/history][item/bat][batnun][volt/curr/temp/coul][item]
disp     Display Info at regular intervals - disp [(pwrs pwrNo)/val]/[(bats batNo)/volt/curr/temp]
getpwr   Get power Info - getpwr
help     Help [cmd]
info     Device infomation - info
log      Log information show - log
login    Login Admin mode - login [password]
logout   user mode  - logout
pwr      Power data show - pwr [index]
shut     Shut down - shut
soh      State of health - soh [addr]
stat     Statistic data show - stat
time     Time - time [year] [month] [day] [hour] [minute] [second]
trst     Test Soft Reset - trst
```

See [this reference](https://gist.github.com/ned-kelly/a56fe92b101190730ac05d81cfc536c4) for command examples that i was able to pull by querying the device directly - happy to provide more if you require also, just let me know.

------------

## Feature Request 😇

Would it be possible to add support for ASCII Console Mode as an option in bms-to-inverter? 🙏

This would allow compatibility with Pylon Extra2000 and older LiFePO4 pylon batteries, which require this initialization step before accepting commands, and from what I can tell don't currently work with the current implementation. A possible minimum viable implementation could include:

1. 	An option to send the console mode activation command (`~20014682C0048520FCC3\r`) before communication begins.
2. 	Switching to ASCII-based commands (pwr, bat, etc.) instead of the current binary protocol.
3. 	Implementing line-based text parsing instead of expecting structured binary frames.

Please let me know if this could be added, or if I can provide any additional details! - PS it's been 15 years since i've written any java, sorry I can't just write this, otherwise i'd submit a PR and have a crack myself, I desperately want to try and get this working with my Growatt SPH 6000 which no longer supports any Lithium battery other than their own as of 2025 :/

ned-kelly, 2025-03-09:
Hi Torsten - I'd be happy to also validate this if #93 is implemented with a python script or something that publishes to a MQTT topic (I'd just need to know all the values required) if that's any help as I see there's a few outstanding issues and I understand how time poor you probably are having my own open source projects that also get inundated with feature requests and issues.

ai-republic, 2025-03-23:
@ned-kelly sorry to answer so late. Been very busy lately and somehow missed this message last week.
Actually this would definitely be possible to integrate this quite easily into the current application. I just need to create a new binding for that and the parsing will be quite some work (not as simple as just reading the value from a data frame), but definitely doable.
Can't promise when I get around to do this as I'll also be quite busy next weekend, but after that it should work out

ned-kelly, 2025-03-23:
Hi Mate, no worries - I’m in a similar boat 😅…

So since I initially posted this, I’ve actually now got a mostly working parsing implantation of this In a python script I’ve actually knocked out in the last few days, however I’m struggling to get the inverter (growatt SPH 6000 - running in rs485 mode) working, and I’ve tried to reverse engineer your implementation without much luck unfortunately - I see you’ve abstracted away the handling of the frames and that’s where I’ve got a bit lost - I’m replying to the inverter when it queries the “growatt battery protocol” using modbus over rs485, however I’m still missing some logic (I suspect some of the frames I’m still not returning the correct format) as it’s throwing errors for me still on the physical inverter.

Anyway, I noticed that you were also considering adding a feature to receive messages for bms-to-inverter via MQTT… in issue #96 - if it helps, I’d be happy to test and get the parsing working properly myself 100% using the MQTT feature (if you’re able to implement it), and then that might make it easier for you to port to logic to the Java implementation? I’ll post up python script, regex etc.

ned-kelly, 2025-05-28:
Just checking if you had any furhter comments on this?