ai-republic/bms-to-inverter #109: Schneider XW Pro

matfew8, 2025-03-12:
Hey stumbed across your thread on the DIY solar forum and had a good read.

Have you looked at the XW pro at all. These have support for a few BMS's you have supported so might be able to make it work.

I have cheap china BMS's atm. I got them and the cells for next to nothing just some labour work so can't complain.

They use the china DR-Jc03 from what I can find.

Any idea how I can make these talk to the XW-Pro inverter. It does have support for a few BMS systems inc Pylon but I am running 16cell batteries not sure if pylon works with 16 or just 15 I have read both on different pages so tad confused.

ai-republic, 2025-03-15:
Hi and welcome @matfew8,
if your XW-Pro supports the Pylon protocol then you should be fine. I've just searched a little and found that the DR-Jc03 also supports the Pylon protocol.
How about your ports? Does the DR-Jc03 support CAN or RS485? And your XW pro inverter?

njwretnm, 2025-04-09:
I think the XW Pro does support Pylon CAN (There are a few Pylontech batteries that are on the XW Pro closed loop support list)  via the Insight Home Gateway (it has CAN on the 12-pin connector): 

https://shop.frankensolar.ca/content/documentation/Pylontech/Appnote_Pylontech_Schneider-Conext-XW-Pro-Li-Ion-Battery-Solution-Guide990-6359-Rev_(frankensolar).pdf

ai-republic, 2025-04-09:
@njwretnm have you tried them with the application? Did it work or did you encounter any issues? Let me know if you need help

njwretnm, 2025-04-09:
I'm waiting for someone to test before I buy :) . I'm considering getting a JK-BMS and slapping something together.  Will reach out when I do!

matfew8, 2025-04-09:
MMM i might have to buy some bits n bobs. I have the XW pro (AUS version) and I do have only 1 JK v15 bms at this stage. Had a battery i acquired with a BMS issue so bought a JK 200A v15 unit for it to play with. 

I'll buy some hardware in next week or so and got a week of work coming up so will get some time to play.

Think I'll be back here looking for some help no doubt haha.

njwretnm, 2025-04-10:
> MMM i might have to buy some bits n bobs. I have the XW pro (AUS version) and I do have only 1 JK v15 bms at this stage. Had a battery i acquired with a BMS issue so bought a JK 200A v15 unit for it [...]

Let us know what you find!  Looking forward to hearing more on this.

mikeiimura, 2025-06-10:
I have a XW Pro 6848 and a Eco-Worthy server rack battery. I tried setting the Insight Home to a Pylontech US5000 and the Eco-Worthy to Pylon and connected them with RS485. It did not work. I suspect that the Schnieider insight is looking for specific battery type string because they have several Pylontech batteries separately listed.

matfew8, 2025-07-08:
Hey, So after a while I have finally got the RP4 running. Got the twin can hat for it and got that configured and working. Found the issue on RP4 java when trying to install but havent got around that yet. I cant seem to get a successful build on a win11 PC or RP4. Any ideas?