ai-republic/bms-to-inverter #114: Narada 48NPFC200

Boglija, 2025-04-04:
Hello, I use Narada 48NPFC100 battery and RS485 CAN HAT as you recommend 
ifconfig 
can0: flags=193<UP,RUNNING,NOARP> mtu 16
inet 192.168.15.15 netmask 255.255.255.0
unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00 txqueuelen 65536 (UNSPEC)
RX packets 28740 bytes 229920 (224.5 KiB)
RX errors 0 dropped 28740 overruns 0 frame 0
TX packets 0 bytes 0 (0.0 B)
TX errors 0 dropped 0 overruns 0 carrier 0  collisions 0
but in my logs 
2025-04-04 16:41:33.638 | INFO | main | verter.BmsToInverter:281 | Starting BMS receiver...
2025-04-04 16:41:33.642 | INFO | Thread-3 | verter.BmsToInverter:307 | Reading BMS #0 NARADA_MODBUS on can0...
2025-04-04 16:41:33.643 | INFO | Thread-3 | toinverter.core.Port:111 | Opening can0...
2025-04-04 16:41:33.665 | ERROR | Thread-3 | dbus.J2ModMasterPort:75 | Could not open modbus master port can0!
com.fazecast.jSerialComm.SerialPortInvalidPortException: Unable to create a serial port object from the invalid port descriptor: /dev/can0
How do I configure can0 correctly?

ai-republic, 2025-04-09:
Hi @Boglija 
you need to use the RS485 port, not the CAN port as the ModBus protocol runs via RS485.
So on you RS485/CAN Hat look for the connectors usually labeled A and B. That will be your RX (receive) and TX (transmit) lines.
Be sure to connect them also correctly on your BMS. If you're certain you have identified the RX and TX wires from your BMS you can test which wire connects to which connector by switching them around.

Boglija, 2025-04-09:
I can provide you with ssh access to raspberi pi

ai-republic, 2025-04-17:
I think your connections or RS485 adapter configuration is not correct as it cannot read from the serial port /dev/ttySC0.
You need to configure the ports baud rate with `stty -F /dev/ttySC0 19200` and also set the same 19200 in the _Configurator_ app when you add/edit a BMS. 
But also double check your connections. Possibly switch the wires A and B on the RS485 adapter.

Boglija, 2025-04-17:
Thank you for your attention. There I have posted the program with which I connect to the BMS. And at the same time I connect to the BMS with your application and I see the BMS responses in the log, so I think that the ports are configured correctly. I have a board with two rc485 ports and if I connect them and send some word from one port, I see it on the second port. Thank you for your help, but I think that I have some kind of BMS that is different from yours.

Boglija, 2025-04-17:
root@HomeAsistant:~# mbpoll -m rtu -a 39 -r 0x1000 -t 4 -b 19200 -p none -s 1 /dev/ttySC1
mbpoll 1.5-2 - ModBus(R) Master Simulator
Copyright (c) 2015-2023 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: ModBus RTU
Slave configuration...: address = [39]
                        start reference = 4096, count = 1
Communication.........: /dev/ttySC1,      19200-8E1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 39... Ctrl-C to stop)
Read output (holding) register failed: Connection timed out
-- Polling slave 39... Ctrl-C to stop)
^C--- /dev/ttySC1 poll statistics ---
2 frames transmitted, 0 received, 1 errors, 100.0% frame loss

everything was closed.
Have a nice day !

ai-republic, 2025-04-18:
> root@HomeAsistant:~# mbpoll -m rtu -a 39 -r 0x1000 -t 4 -b 19200 -p none -s 1 /dev/ttySC1 mbpoll 1.5-2 - ModBus(R) Master Simulator Copyright (c) 2015-2023 Pascal JEAN, [...]

This prograom was not able to send and receive anything `2 frames transmitted, 0 received, 1 errors, 100.0% frame loss` so I still think there is some port configuration or wiring issue.

Boglija, 2025-04-18:
[image]

Boglija, 2025-04-18:
although I also have problems with connecting the powmr 6.2k inverter there is no connection via the PYLON_RS485 protocol with the battery simulator plugin enabled

Boglija, 2025-04-18:
[image]
This is what I get in the console if I connect the monitoring program to Windows

Boglija, 2025-04-18:
[image]

ai-republic, 2025-04-19:
The pictures above don't say much about the protocol.
Your connections seems to work going from your RS485 hat A / B -> B / A. But do you get readings from your BMS via the RS485 hat?

Boglija, 2025-04-19:
No, there is no information from the battery. It seems to me that you need to ask her something to get an answer.

Boglija, 2025-04-19:
I think that my bms is not the same as yours. I have some other one like v-tac vt48100e-w ess

Boglija, 2025-04-22:
The representatives of v-tac answered me, I don’t know if this information will help

[Growatt OffGrid SPF Modbus RS485&RS232 RTU Protocol V0.12  20200724A.pdf](https://github.com/user-attachments/files/19849717/Growatt.OffGrid.SPF.Modbus.RS485.RS232.RTU.Protocol.V0.12.20200724A.pdf)

[Voltronic Inverter and BMS 485 communication protocol 20200325（KINGVMIIIVMIVMKSIIIMKSIVMAXMAXII）.pdf](https://github.com/user-attachments/files/19849719/Voltronic.Inverter.and.BMS.485.communication.protocol.20200325.KINGVMIIIVMIVMKSIIIMKSIVMAXMAXII.pdf)

[【phocos】Any-Grid_PSW-H_RS-485 MODBUS RTU Battery Protocol_2021-04-07.pdf](https://github.com/user-attachments/files/19849720/phocos.Any-Grid_PSW-H_RS-485.MODBUS.RTU.Battery.Protocol_2021-04-07.pdf)

ai-republic, 2025-05-02:
@Boglija thanks for the specs!
the Any-Grid_PWS and Voltronic look very similar. Voltronic provides more data but the key values are the same.
Growatt is completely different though. And also doesn't match the current _GROWATT_MODBUS_ binding.
Looks like I got some more bindings to implement 😉 
I would go with the Any-Grid_PWS first as its a subset of the Voltronic spec.

Boglija, 2025-05-02:
TX:(2025-05-02 06:35:22)，Sent【6】Bytes
COM7:7C 01 01 00 02 0D 
RX:(2025-05-02 06:35:22)，Received【134】Bytes
7C 01 01 80 01 10 8D 49 0D 4B 0D 4B 8D 53 0D 49 0D 4C 8D 50 0D 4E 8D 4A 0D 4E 8D 4A 0D 4D 8D 4D 0D 4A 8D 4A 0D 3C 02 01 73 54 03 01 26 84 04 01 27 10 05 06 00 46 00 46 00 46 00 46 40 49 20 48 06 05 00 00 00 01 00 00 00 00 00 00 07 01 00 08 08 01 15 44 09 01 27 10 0A 01 00 00 0B 01 00 00 02 F1 0C 01 00 00 02 4E 0D 01 00 04 11 CC 0E 01 00 02 F8 50 0F 01 00 00 98 BC 10 01 00 00 76 C0 11 01 00 00 FC 0D 

TX:(2025-05-02 06:35:23)，Sent【6】Bytes
COM7:7C 01 01 00 02 0D 
RX:(2025-05-02 06:35:23)，Received【134】Bytes
7C 01 01 80 01 10 8D 49 0D 4B 0D 4C 8D 52 0D 4A 0D 50 8D 4D 0D 50 8D 49 0D 4E 8D 4A 0D 4F 8D 4D 0D 4A 8D 4A 0D 40 02 01 73 43 03 01 26 84 04 01 27 10 05 06 00 46 00 46 00 46 00 46 40 49 20 48 06 05 00 00 00 01 00 00 00 00 00 00 07 01 00 08 08 01 15 45 09 01 27 10 0A 01 00 00 0B 01 00 00 02 F1 0C 01 00 00 02 4E 0D 01 00 04 11 CC 0E 01 00 02 F8 50 0F 01 00 00 98 BC 10 01 00 00 76 C0 11 01 00 00 72 0D 

TX:(2025-05-02 06:35:24)，Sent【6】Bytes
COM7:7C 01 01 00 02 0D 
RX:(2025-05-02 06:35:24)，Received【134】Bytes
7C 01 01 80 01 10 8D 49 0D 4B 0D 4B 8D 52 0D 4A 0D 4C 8D 50 0D 4F 8D 4A 0D 4F 8D 4A 0D 4E 8D 4D 0D 4C 8D 4A 0D 3E 02 01 73 4E 03 01 26 84 04 01 27 10 05 06 00 46 00 46 00 46 00 46 40 49 20 48 06 05 00 00 00 01 00 00 00 00 00 00 07 01 00 08 08 01 15 45 09 01 27 10 0A 01 00 00 0B 01 00 00 02 F1 0C 01 00 00 02 4E 0D 01 00 04 11 CC 0E 01 00 02 F8 50 0F 01 00 00 98 BC 10 01 00 00 76 C0 11 01 00 00 E4 0D 

TX:(2025-05-02 06:35:25)，Sent【6】Bytes
COM7:7C 01 01 00 02 0D 
RX:(2025-05-02 06:35:25)，Received【134】Bytes
7C 01 01 80 01 10 8D 49 0D 4B 0D 4B 8D 52 0D 49 0D 4F 8D 4E 0D 50 8D 49 0D 4F 8D 4A 0D 4F 8D 4D 0D 4A 8D 4A 0D 40 02 01 73 61 03 01 26 84 04 01 27 10 05 06 00 46 00 46 00 46 00 46 40 49 20 48 06 05 00 00 00 01 00 00 00 00 00 00 07 01 00 08 08 01 15 45 09 01 27 10 0A 01 00 00 0B 01 00 00 02 F1 0C 01 00 00 02 4E 0D 01 00 04 11 CC 0E 01 00 02 F8 50 0F 01 00 00 98 BC 10 01 00 00 76 C0 11 01 00 00 AC 0D 

TX:(2025-05-02 06:35:26)，Sent【6】Bytes
COM7:7C 01 01 00 02 0D 
RX:(2025-05-02 06:35:26)，Received【134】Bytes
7C 01 01 80 01 10 8D 48 0D 4B 0D 4B 8D 53 0D 49 0D 4C 8D 50 0D 4E 8D 4A 0D 4E 8D 4A 0D 4E 8D 4D 0D 4A 8D 49 0D 3C 02 01 73 72 03 01 26 84 04 01 27 10 05 06 00 46 00 46 00 46 00 46 40 49 20 48 06 05 00 00 00 01 00 00 00 00 00 00 07 01 00 08 08 01 15 44 09 01 27 10 0A 01 00 00 0B 01 00 00 02 F1 0C 01 00 00 02 4E 0D 01 00 04 11 CC 0E 01 00 02 F8 50 0F 01 00 00 98 BC 10 01 00 00 76 C0 11 01 00 00 F8 0D 

TX:(2025-05-02 06:35:27)，Sent【6】Bytes
COM7:7C 01 01 00 02 0D 
RX:(2025-05-02 06:35:27)，Received【134】Bytes
7C 01 01 80 01 10 8D 48 0D 4B 0D 4C 8D 52 0D 4A 0D 50 8D 4E 0D 50 8D 49 0D 4E 8D 4A 0D 50 8D 4D 0D 4C 8D 49 0D 42 02 01 73 69 03 01 26 84 04 01 27 10 05 06 00 46 00 46 00 46 00 46 40 49 20 48 06 05 00 00 00 01 00 00 00 00 00 00 07 01 00 08 08 01 15 46 09 01 27 10 0A 01 00 00 0B 01 00 00 02 F1 0C 01 00 00 02 4E 0D 01 00 04 11 CC 0E 01 00 02 F8 50 0F 01 00 00 98 BC 10 01 00 00 76 C0 11 01 00 00 9A 0D

ai-republic, 2025-05-03:
Hmm, this does not match any of the specifications. The specifications say nothing about function code 0x01 (third value)

Boglija, 2025-05-03:
I opened the battery there is a vestwoods bms I wrote to vestwoods they gave me a program for windows on the main page there are logs of communication with the battery which I sent to you They can't give you anything else

Boglija, 2025-05-03:
But the program has some very interesting features, I'll send you a photo now

Boglija, 2025-05-03:
[image]

Boglija, 2025-05-03:
[image]

Boglija, 2025-05-03:
[image]

Boglija, 2025-05-03:
I'm afraid to choose something and then it might not connect.

Boglija, 2025-05-03:
[image]

ai-republic, 2025-05-04:
Ok, that looks promising. But as your Narada 48NPFC100 only supports RS485 (and you have the RS485 hat) it would be interesting what RS485 protocols it supports. Maybe you can send a screenshot of the RS485 configuration options?

Boglija, 2025-05-04:
I don't know if I understood you correctly.

[image]

ai-republic, 2025-05-04:
I was thinking more on the line what protocol options are under the RS485 configuration here:

[image]

Boglija, 2025-05-04:
this is all

Boglija, 2025-05-08:
[123.txt](https://github.com/user-attachments/files/20110343/123.txt)
python3 123.py
Сирі значення: [31745, 384, 272, 3360, 3361, 3362, 3359, 3361, 3363, 3359, 3359, 3360, 3360, 3360, 3362, 3364, 3360, 3361, 3360, 513, 30000, 769, 10000, 1025, 10000, 1286, 65, 65, 65, 65, 16451, 8258, 1541, 0, 0, 0, 0, 0, 1793, 8, 2049, 5377, 2305, 10000, 2561, 0, 2817, 0, 757, 3073, 0, 590, 3329, 4, 7196, 3585, 2, -1968, 3841, 0, -26236, 4097, 0, 30400, 4353, 0, 27149]
🔋 **Дані батареї:**
Напруга комірок: [3.36, 3.361, 3.362, 3.359, 3.361, 3.363, 3.359, 3.359, 3.36, 3.36, 3.36, 3.362, 3.364, 3.36, 3.361, 3.36]
Максимальна напруга: 3.364 В
Мінімальна напруга: 3.359 В
⚖️ Дисбаланс комірок (Макс - Мін): 0.005 В
🌡️ Температура BMS: 16.45°C
🔄 Кількість циклів заряду: 8
⚡ Струм заряду: 0.00 A
⚡ Струм розряду: 0.00 A

ai-republic, 2025-05-17:
hmm, can't make much sense of the values except for these: 3360, 3361, 3362, 3359, 3361, 3363, 3359, 3359, 3360, 3360, 3360, 3362, 3364, 3360, 3361, 3360. These will be the cell voltages

closed 2026-01-30 by ai-republic