ai-republic/bms-to-inverter #121: Support for Easun SMH-II-3.2K

jlmontes1697, 2025-04-16:
I have an Easun SMH-II 3.2K inverter (Voltronic compatible, I believe) with an RS232 connection and a Daly 7S Lipo BMS.
The bms is working good but can't connect the Java to the inverter.

ai-republic, 2025-04-17:
Hi @jlmontes1697,
from the Easun SMH-II 3.2K datasheet I see that it supports RS485 so that should be fine.
I'm not sure what protocol the inverter supports but I would guess the Pylon as most chinese inverters use/support that.
Have you tried the BMS-to-Inverter application? What is your setup?

jlmontes1697, 2025-04-27:
The inverter documentation says it has RS485/RS232. Later in the documentation, it says it only has RS232. I haven't been able to read anything via RS485, so it only works with RS232, and I don't see that you have implemented the P30 protocol in the code.

ai-republic, 2025-05-02:
Can you provide the P30 protocol communication specification?