ai-republic/bms-to-inverter #124: Data from HomeAssistant

whitoSB, 2025-04-28:
Hi, thanks for your work, I'm new to this, but I was wondering if it would be possible to send data from HA to the inverter? I have a victron smartshunt and ANT BMS via bluetooth inserted in HA. Would I be able to send values ​​via pylon rs485 to the easun smx II? Thank you for your time and answer

ai-republic, 2025-05-02:
How would you send data from HA to the inverter? Via a MQTT bridge? Like HA sends voltage, current, limits via MQTT to the application and then to the inverter?
Yes, I believe that could be possible.

whitoSB, 2025-05-02:
yes, that's exactly what I meant

ai-republic, 2025-05-02:
Ok, I'm not sure when I will get around to that - I'm quite busy at the moment. But it shouldn't take to long as there's already a MQTT server and client that reads the values (previously for the webserver).

whitoSB, 2025-05-02:
It's okay, I understand. Thank you for your time.

whitoSB, 2025-05-03:
Today I had time and I tried to track what requests the inverter sends via rs485: these 3 strings are repeated:

7E 32 30 31 32 34 36 36 31 30 30 30 30 46 44 41 41 0D
7E 32 30 31 32 34 36 36 32 30 30 30 30 46 44 41 39 0D
7E 32 30 31 32 34 36 36 33 30 30 30 30 46 44 41 38 0D

easun smx II inverter set to BMS and PYL protocol, inverter has error 58, BMS not connected, I tried to simulate BMS using your method, but the inverter does not respond. I am new to this, so I may be making a mistake somewhere.

ai-republic, 2025-05-04:
Could you attach a log file please?

whitoSB, 2025-05-04:
[BMS-to-Inverter_9600.log](https://github.com/user-attachments/files/20027868/BMS-to-Inverter_9600.log)

ai-republic, 2025-05-12:
So the inverter sends these requests continuously?
7E 32 30 31 32 34 36 36 31 30 30 30 30 46 44 41 41 0D  -> request analog battery info
7E 32 30 31 32 34 36 36 32 30 30 30 30 46 44 41 39 0D  -> request system status / alarms
7E 32 30 31 32 34 36 36 33 30 30 30 30 46 44 41 38 0D  -> request dis-/charge info

If that is the case, I need to rework to answer to the requests and not actively send the data (responses)

whitoSB, 2025-05-12:
yes, the inverter is sending these three requests, thank you for your time

ai-republic, 2025-05-17:
Alright, I have change the binding for this to read and reply to those request. I don't have time on this weekend, but maybe next

whitoSB, 2025-05-17:
I understand, thank you

ai-republic, 2025-07-04:
Hi @it-koncept,
no, sorry, didn't have any time yet

closed 2026-10-01 by whitoSB