ai-republic/bms-to-inverter #125: Configurator.jar resulting in error exit

dingoduffel, 2025-05-04:
Hi,
I have been executing configurator.jar on my Pi 2 B (running both java and javac version 1.8.0_202) with the following result:

[INFO] bms-to-inverter-parent ............................. SUCCESS [  3.399 s]
[INFO] core-api-0.0.1-SNAPSHOT ............................ FAILURE [  8.403 s]
[INFO] protocol-can-0.0.1-SNAPSHOT ........................ SKIPPED
[INFO] bms-pylon-hv-can-0.0.1-SNAPSHOT .................... SKIPPED
[INFO] inverter-sma-si-can-0.0.1-SNAPSHOT ................. SKIPPED
[INFO] webserver-0.0.1-SNAPSHOT ........................... SKIPPED
[INFO] bms-to-inverter-main-0.0.1-SNAPSHOT ................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  13.919 s
[INFO] Finished at: 2025-05-03T14:12:19+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.10.1:compile (default-compile) on project core-api: Fatal error compiling: invalid flag: --release -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :core-api

Failed to build application!

So it seems the maven compiler has been called with an invalid flag --release. Now it seems the --release flag has been only introduced from javac 9 onwards?
Anybody else encountered this and got it solved?

dingoduffel, 2025-05-04:
OK, so issue solved and closed:
The suggestion to install Java JDK 8 with an 32 bit OS is not working (--release flag being only supported from release 9 onwards)
So I downloaded https://downloads.raspberrypi.com/raspios_full_armhf/images/raspios_full_armhf-2024-11-19/2024-11-19-raspios-bookworm-armhf-full.img.xz and flashed it with Raspberry Pi Imager, which - according to the release notes contains a more recent version of Java(c).
And this did the trick. So I suggest the Wiki How it works to be updated pointing "poor" 32-bit Pi owners to the Bookworm OS with recommended software and reflash their boot disk

closed 2025-05-05 by dingoduffel

ai-republic, 2025-05-12:
Thanks for the info - I'll look into that. Should be compilable without the --release flag. Possibly need to downgrade Maven version. I am compiling with JDK8 and Maven 3.9.6 locally without problems. The same Maven version is downloaded during update/installation using the _Configurator_ app.

ai-republic, 2025-05-12:
Oh, just came across this discussion post: [Installation on Rasperian - Failed to install the application](https://github.com/ai-republic/bms-to-inverter/discussions/129)
That might help your JDK 8 issue.