ai-republic/bms-to-inverter #126: How to use SimulatedBatteryPackPlugin?

Citizen4our, 2025-05-04:
Hello, 
I'm trying to simulate a battery for an inverter using the PYLON_RS485 protocol, but I'm getting zero values for all parameters.
Whats going wrong? how to use  plugin(s)?
My configuration settings are shown in the screenshots below:

[image]

[image]

[image]

stdoutput:

[image]

[image]

my config:

```
###################################################################
###                  System specific settings                   ###
###################################################################

###################################################################
###                       BMS settings                          ###
###################################################################

####  Simple single port configuration ####
# bms.pollIntervall - is the interval to request BMS data (in seconds)
# bms.x.type - can be (DALY_CAN, DALY_RS485, JK_CAN, PYLON_CAN or SEPLOS_CAN 
# bms.x.portLocator - is the locator/device to use to communicate to the BMS, eg. can0, /dev/ttyUSB0, com3, etc.  
# bms.x.baudRate - is the locator/device baudrate to use to communicate to the BMS, eg. 9600, 500000, etc.  
# bms.x.delayAfterNoBytes - is the delay after receiving no data (in ms)
bms.pollInterval=1

bms.1.type=NONE
bms.1.id=1
bms.1.portLocator=can1
bms.1.baudRate=500000
bms.1.delayAfterNoBytes=200

###################################################################
###                    Inverter settings                        ###
###################################################################
# The inverter type can be NONE, DEYE_CAN, GROWATT_CAN, SMA_SI_CAN, SOLARK_CAN
inverter.type=PYLON_RS485
# The port name/device to use to communicate to the  inverter  
inverter.portLocator=/dev/ttys005
# The port baud rate to use to communicate to the  inverter  
inverter.baudRate=9600
# Interval to send data to the inverter (in seconds)
inverter.sendInterval=1

###################################################################
###                     Plugin settings                         ###
###################################################################
plugin.inverter.1.class=com.airepublic.bmstoinverter.core.plugin.inverter.SimulatedBatteryPackPlugin
plugin.inverter.1.property.1.name=SOC
plugin.inverter.1.property.1.value=500
plugin.inverter.1.property.1.description=The configured preset batterypack SOC (unit 0.1%)
plugin.inverter.1.property.2.name=SOH
plugin.inverter.1.property.2.value=990
plugin.inverter.1.property.2.description=The configured preset batterypack SOH (unit 0.1%)
plugin.inverter.1.property.3.name=Current
plugin.inverter.1.property.3.value=100
plugin.inverter.1.property.3.description=The configured preset batterypack current (unit 0.1A)
plugin.inverter.1.property.4.name=Voltage
plugin.inverter.1.property.4.value=520
plugin.inverter.1.property.4.description=The configured preset batterypack voltage (unit 0.1V)
plugin.inverter.1.property.5.name=Max. charge current
plugin.inverter.1.property.5.value=200
plugin.inverter.1.property.5.description=The configured preset batterypack maximum charge current (unit 0.1A)
plugin.inverter.1.property.6.name=Max. discharge current
plugin.inverter.1.property.6.value=200
plugin.inverter.1.property.6.description=The configured preset batterypack maximum discharge current (unit 0.1A)
plugin.inverter.1.property.7.name=Max. voltage limit
plugin.inverter.1.property.7.value=540
plugin.inverter.1.property.7.description=The configured preset batterypack maximum voltage limit (unit 0.1V)
plugin.inverter.1.property.8.name=Min. voltage lime
plugin.inverter.1.property.8.value=480
plugin.inverter.1.property.8.description=The configured preset batterypack minimum voltage limit (unit 0.1V)
plugin.inverter.1.property.9.name=Average Temperature
plugin.inverter.1.property.9.value=250
plugin.inverter.1.property.9.description=The configured preset batterypack average temperature (unit 0.1C)

plugin.inverter.2.class=com.airepublic.bmstoinverter.core.plugin.inverter.ManipulateBatteryPackDataPlugin
plugin.inverter.2.property.1.name=SOC
plugin.inverter.2.property.1.value=
plugin.inverter.2.property.1.description=The configured preset batterypack SOC (unit 0.1%)
plugin.inverter.2.property.2.name=SOH
plugin.inverter.2.property.2.value=
plugin.inverter.2.property.2.description=The configured preset batterypack SOH (unit 0.1%)
plugin.inverter.2.property.3.name=Current
plugin.inverter.2.property.3.value=
plugin.inverter.2.property.3.description=The configured preset batterypack current (unit 0.1A)
plugin.inverter.2.property.4.name=Voltage
plugin.inverter.2.property.4.value=
plugin.inverter.2.property.4.description=The configured preset batterypack voltage (unit 0.1V)
plugin.inverter.2.property.5.name=Max. charge current
plugin.inverter.2.property.5.value=
plugin.inverter.2.property.5.description=The configured preset batterypack maximum charge current (unit 0.1A)
plugin.inverter.2.property.6.name=Max. discharge current
plugin.inverter.2.property.6.value=
plugin.inverter.2.property.6.description=The configured preset batterypack maximum discharge current (unit 0.1A)
plugin.inverter.2.property.7.name=Max. voltage limit
plugin.inverter.2.property.7.value=
plugin.inverter.2.property.7.description=The configured preset batterypack maximum voltage limit (unit 0.1V)
plugin.inverter.2.property.8.name=Min. voltage lime
plugin.inverter.2.property.8.value=
plugin.inverter.2.property.8.description=The configured preset batterypack minimum voltage limit (unit 0.1V)
plugin.inverter.2.property.9.name=Average Temperature
plugin.inverter.2.property.9.value=
plugin.inverter.2.property.9.description=The configured preset batterypack average temperature (unit 0.1C)

###################################################################
###                 Optional services settings                  ###
###################################################################

#### Webserver properties ####
webserver.service.enabled=true
# The webserver port
webserver.http.port=8080
webserver.https.port=8443
webserver.username=
webserver.password=
```

ai-republic, 2025-05-12:
Hmm, looks like you have 2 plugins configured. Whereas you set the values in the _SimulatedBatteryPackPlugin_ I think you delete them in the _ManipulateBatteryPackDataPlugin_

Citizen4our, 2025-05-12:
Hi @ai-republic 
with 1 SimulatedBatteryPackPlugin have same result

[image]

[image]

ai-republic, 2025-05-17:
Ah, I think this could be related to issue #124 where the inverter sends the requests and expects a reply and not to actively send the data to the inverter. I'll try and fix that next weekend as I don't have time until then