ai-republic/bms-to-inverter #127: add support for SMA hybrid Sunny tripower

dingoduffel, 2025-05-09:
I tried the SMA Sunny Island inverter protocol on my SMA hybrid Sunny tripower (with a dummy BMS), but this doesn't seem to work.
I'm adding here the trace of the telegrams that are sent by the SMA. Hope this makes sense to someone, as well as it shows the difference between the Sunny Island and the Sunny Tripower

[inverter.txt](https://github.com/user-attachments/files/20121081/inverter.txt)

ai-republic, 2025-05-12:
Yes, it seems the SMA Tripower uses a different protocol than the Sunny Island. Maybe you can get your hand on it so I can integrate it?

dingoduffel, 2025-05-12:
I found an implementation of the SMA hybrid protocol on https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-SMA-Sunny-Tripower-Smart-Energy.
I contacted Dalathegreat and this implementation was obtained by reverse engineering an existing installation. In the code https://github.com/dalathegreat/Battery-Emulator/blob/main/Software/src/inverter/SMA-TRIPOWER-CAN.cpp you can seehow they implemented the protocol. But on top, you would need an extra connection for the "enable" hardware signal, which is supposed to cutoff the battery. So either this signal is imported on a GPIO of a PI (or alike) but no longer universally implementable, or - what I will look at - an external actuator that cuts the battery power.

dingoduffel, 2025-05-12:
Hi Torsten,

pls see my answer in the ticket.

brgds,

Dirk

ratio91, 2025-05-12:
I also own a SMA Sunny Tripower SE (and need to get it working with a Pylontech HV Battery) - so let me know if I can help.

dingoduffel, 2025-05-12:
@ratio91, from the datasheet of the Pylontech I can see you will face a similar issue: the battery has no input pin to enable/disable the battery, which the SMA requires. For that reason I have now ordered an external contactor that will interrupt the battery voltage to the inverter (based on the "enable" signal that is on the I/O connector of the inverter) I ordered today the TE Connectivity KILOVAC LEV100 Contactor with 12 V dc Coil (e.g.https://export.rsdelivers.com/product/te-connectivity/lev100a5anh-4-1618391-1/te-connectivity-kilovac-kilovac-lev100-contactor-v/2777222) but until the protocol is implemented in the bms-to-inverter, I won't be able to tell if this contactor is doing its job

ratio91, 2025-05-12:
thanks for the detailed instructions @dingoduffel - for now I run my PylonTech batteries with a separate Goodwe GW8K-ET (which I got for about 500 EUR) since I had a hard time connecting it to the SMA inverter. This obviously leads to some efficiency losses, as DC-AC-DC conversion is happening. SO ideally, at some point I can get rid of the GoodWe inverter and use my existing SMA DC-DC with my batteries.

closed 2026-01-30 by ai-republic