ai-republic/bms-to-inverter #128: Add BLE Support?

admiralroflknife, 2025-05-10:
I'm trying to connect dchouse golf cart batteries that have a jbd bms in them which offer a BLE connection for BMS data. 

I have 2 ecoworthy 5000w inverters (srne hyp4850 rebadged) that support multiple battery protocols, so would work with this libary easily. 

I'm stuck because I was looking at using Yambus because it supports the BLE side for data input, but its output is only CAN. You library already works for rs485, but lacks BLE support.

is it possible to add BLE data support?

ai-republic, 2025-05-12:
I guess that would be nice but I only have the weekends to work of some off the issues. Its definitely something to add in the future.
If the JBD support bluetooth they should also support serial communication like RS485. Have you checked that?

admiralroflknife, 2025-05-12:
> I guess that would be nice but I only have the weekends to work of some off the issues. Its definitely something to add in the future. If the JBD support bluetooth they should also support serial communication like RS485. Have you checked that?

These are golf cart style batteries sealed in plastic cases, so the Bms is not accessible without prying the box apart, so that’s why the aim towards using a Bluetooth aspect of them. 

They were a good deal at the time, with hindsight I wish I had sprung the extra for the server rack style batteries but so be it. 

I use this library currently to import my bms data into home assistant 
https://github.com/syssi/esphome-jbd-bms