ai-republic/bms-to-inverter #135: AXIOMA-ISGRID-HV-10000 inverter with SOLAX Triple Power 3.0 battery

Oles0968, 2025-06-20:
Hi @ai-republic,

First of all, thank you for your project; I hope I will be successful in my case.
So, I'm starting my long read ;)

I have an AXIOMA-ISGRID-HV-10000 inverter, which is a clone of the Deye SUN-10K-SG01HP3-EU-AM2 and SOLAX Triple Power 3.0 T-BAT H 6.0, with SOLAX BMS MC0600. According to a request to the AXIOMA service center, the inverter is not compatible with this battery manufacturer.

Both are using the CAN protocol.
The inverter connects to the BMS and can read data.

[image]

I decided I needed your support because:

1. The inverter sees the wrong battery capacity value – 50 Ah, while the real value is 30 Ah
2. They work together for about +/- 1 hour, then the inverter receives an alarm signal F56 BAT1_VoltLow_Fault, and the BMS shuts down the battery.

[image]

So, I assumed that not all the data was being transmitted correctly.

I assembled the Raspberry Pi 5 with the Waveshare 2-Channel CAN FD HAT, connected the RJ45s based on the inverter and BMS documentation.
Inverter, port can1
[image]

BMS, port can0
[image]

Raspberry Pi 5 with the Waveshare 2-Channel CAN FD HAT
[image]

In Configurator, I choose the Deye inverters (CAN) protocol for the inverter and PYLON_HV_CAN for BMS, as I read that SOLAX HV inverters support this protocol. Unfortunately, I do not have official documentation for the SOLAX protocol.
After connecting all the components and starting the battery with the inverter, I received an alarm F58: BMS communication error.

I attached the log file.

[BMS-to-Inverter.log](https://github.com/user-attachments/files/20835188/BMS-to-Inverter.log)

I will be happy if you can help me.

[Deye-Approved-Battery-list_DY_HV_160_800_025_.pdf](https://github.com/user-attachments/files/20835320/Deye-Approved-Battery-list_DY_HV_160_800_025_.pdf)

[ISGRID-HV_10-15-20_Datasheet-AXIOMAenergy-Translated.pdf](https://github.com/user-attachments/files/20835409/ISGRID-HV_10-15-20_Datasheet-AXIOMAenergy-Translated.pdf)

[t-bat-sys-hv-3.0-datasheet-en.pdf](https://github.com/user-attachments/files/20835414/t-bat-sys-hv-3.0-datasheet-en.pdf)

Oles0968, 2025-07-04:
Hi @ai-republic,

I would be very grateful for any suggestions regarding my case.

ai-republic, 2025-07-05:
Hi @Oles0968,
sorry, I didn't see you initial post. 
It looks like you're running in log-level info. Could you please configure log-level DEBUG so that I can see more information on what data is being sent and received.
Please attach the BMS-to-Inverter.log and maybe just configure the BMS first and set the inverter to NONE.

ai-republic, 2025-07-11:
Hi @Oles0968,
I think you you didn't pull up the CAN ports before testing (or your configuration/wiring is not correct). 
It cannot send anything to the CAN port (_'Network is down'_ ) to start receiving data from your BMS.

[image]

First try to re-initialized your CAN ports. I made a little `startCAN.sh` script that does this at startup:
```
#!/usr/bin/env bash
sudo ip link set down can0
sudo ip link set down can1
sudo ip link set can0 type can bitrate 500000 restart-ms 100 fd off
sudo ip link set can1 type can bitrate 500000 restart-ms 100 fd off
sudo ifconfig can0 txqueuelen 65536
sudo ifconfig can1 txqueuelen 65536
sudo ip link set up can0
sudo ip link set up can1
```
Then please try again. But please also disable the webserver service as it puts a lot of clutter in the log. Let's just try and get data from your BMS first 😉

Oles0968, 2025-07-11:
Hi @ai-republic,

Thank you, you are right, I forgot to initialize the CAN ports.
My next attempt showed an error on the screen; if I understood correctly, something is wrong with my physical connections.

[image]

So I need to understand where my fault is.

Oles0968, 2025-08-20:
Hi @ai-republic,

Thank you for your involvement in my issues.
I don't have a RPi4B for now, but I found a solution that satisfies me.
I would like to share, maybe it helps someone who has AXIOMA-ISGRID-HV-10000 inverter and SOLAX Triple Power 3.0 T-BAT H 6.0, with SOLAX BMS MC0600.
I can’t say that the issue disappeared completely. Sometimes, the inverter gets an error, but the battery restarts automatically, which makes my life a little happy 😃 
I changed the Energy pattern checkbox to the Batt First (before was Load First).

[image]

closed 2025-08-20 by Oles0968