ai-republic/bms-to-inverter #138: Require help to configure an SD card

jakeyduke, 2025-07-12:
I currently have 8 Samsung SDI ELMP482-00005 batteries nearing their lifespan as im unable to configure the application using my raspberry pie 4b with a waveshare 2 channel can hat in order to have them receive any type of charging current from my Deye 6kw hybrid inverter, as the BMS is not allowing a charge current on open loop. There is one of four LEDs bar of charge remaining in each module.
I'm really hoping there is someone out there who could possibly configure an SD card containing all the information required to have my inverter communicate with my Samsung modules, plug and play if you will. I'm not skilled with any type of programming knowledge and have tried painstakingly over and over again for the past few months to do administer this configuration with my poor computing illiterate skills with no avail. I will be overly appreciative if anyone could help me out as my total investment in my setup is close to 10k now and is about to be worthless ( apart from my inverter) if I don't receive the assistance required to have my system at very least accept a charge.
I'm willing to pay up to $500 to any kind soul out there who is willing to configure this SD card for me as a last ditch attempt to recoup my financial investment and sanity. Any help what so ever would be greatly appreciated and would mean the world to me at this point in time. Thanks 🙏

Rojeshkhatri, 2025-07-31:
@jakeyduke  If you still have issue reachout to me xrider.xr@outlook.com,  i may be able to assist you with the setup.

ai-republic, 2025-08-09:
@jakeyduke were you able to help him set up his system?
If not, then you can contact me also. I'd have to buy a SD card and need your address. No guarantee that it will work out-of-the-box immediately, but if the batteries BMS is compatible with any of the protocols I'm sure we can get it jumping

ai-republic, 2025-08-15:
Best Just to send me you email address here and then I'll delete the post. I'll send you my contact details

jakeyduke, 2025-08-15:
My email address is ***@***.***

jakeyduke, 2025-08-17:
Sorry mate, jake_brunsdon#hotmail.com

Well, that didn't work 😆
Try to leave the @ out of the address and just replace it with #

ai-republic, 2025-08-25:
Hi @jakeyduke,
I've got everything ready for you 😃 
I bought a 16GB  SanDisk SD card ($13). Had quite some trouble getting my RPi image cloned and shrunk to size to fit but after some research and testing got it all running
Also I changed the user and configuration to run as default user (solarpi) its all working now 😃 

I tested the setup on my own RPi 4B and added the scripts to auto start everything on start-up.

I also researched and found the CAN communication specification for the Samsung SDI ELMP482-00005 batteries.

So now I configured the system for your 8 Samsung SDI ELMP482-00005 batteries (ID 1 to ID 8) on `can0` and also to use Deye as inverter on `can1`.

Basically it should be plug-n-play once you insert the SD card into your RPi and boot 😉 

I'll send you an email later for your address. 
You can use the _Donation_ button to cover expenses and donation 😉 ❤️

ai-republic, 2025-08-26:
@jakeyduke I sent you an email