ai-republic/bms-to-inverter #139: Daly can cannot compile using the configurator

svajuss, 2025-07-13:
[image]

Cannot pick daly can communication protocol always getting an error i can only use rs485 which is working perfectly but when i try to send the info to inverter i geting invalid sent frames

ai-republic, 2025-07-18:
Please attach a log file where with debug level where only the BMSes and inverter is configured

svajuss, 2025-07-18:
Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.263 | INFO  | Thread-3   | n.DalyMessageHandler:309   | Battery #2 voltages:

Jul 13 12:50:50 raspberrypi bash[640]:         #1: 3.711V

Jul 13 12:50:50 raspberrypi bash[640]:         #2: 3.714V

Jul 13 12:50:50 raspberrypi bash[640]:         #3: 3.713V

Jul 13 12:50:50 raspberrypi bash[640]:         #4: 3.728V

Jul 13 12:50:50 raspberrypi bash[640]:         #5: 3.727V

Jul 13 12:50:50 raspberrypi bash[640]:         #6: 3.728V

Jul 13 12:50:50 raspberrypi bash[640]:         #7: 3.709V

Jul 13 12:50:50 raspberrypi bash[640]:         #8: 3.71V

Jul 13 12:50:50 raspberrypi bash[640]:         #9: 3.711V

Jul 13 12:50:50 raspberrypi bash[640]:         #10: 3.71V

Jul 13 12:50:50 raspberrypi bash[640]:         #11: 3.709V

Jul 13 12:50:50 raspberrypi bash[640]:         #12: 3.71V

Jul 13 12:50:50 raspberrypi bash[640]:         #13: 3.714V

Jul 13 12:50:50 raspberrypi bash[640]:         #14: 3.714V

Jul 13 12:50:50 raspberrypi bash[640]:         #15: 3.715V

Jul 13 12:50:50 raspberrypi bash[640]:         #16: 3.701V

Jul 13 12:50:50 raspberrypi bash[640]:         #17: 3.705V

Jul 13 12:50:50 raspberrypi bash[640]:         #18: 3.704V

Jul 13 12:50:50 raspberrypi bash[640]:         #19: 3.712V

Jul 13 12:50:50 raspberrypi bash[640]:         #20: 3.707V

Jul 13 12:50:50 raspberrypi bash[640]:         #21: 3.709V

Jul 13 12:50:50 raspberrypi bash[640]:         #22: 3.701V

Jul 13 12:50:50 raspberrypi bash[640]:         #23: 3.699V

Jul 13 12:50:50 raspberrypi bash[640]:         #24: 3.702V

Jul 13 12:50:50 raspberrypi bash[640]:         #25: 3.71V

Jul 13 12:50:50 raspberrypi bash[640]:         #26: 3.709V

Jul 13 12:50:50 raspberrypi bash[640]:         #27: 3.711V

Jul 13 12:50:50 raspberrypi bash[640]:         #28: 3.713V

Jul 13 12:50:50 raspberrypi bash[640]:         #29: 3.713V

Jul 13 12:50:50 raspberrypi bash[640]:         #30: 3.713V

Jul 13 12:50:50 raspberrypi bash[640]:         #31: 3.694V

Jul 13 12:50:50 raspberrypi bash[640]:         #32: 3.695V

Jul 13 12:50:50 raspberrypi bash[640]:         #33: 3.691V

Jul 13 12:50:50 raspberrypi bash[640]:         #34: 3.716V

Jul 13 12:50:50 raspberrypi bash[640]:         #35: 3.716V

Jul 13 12:50:50 raspberrypi bash[640]:         #36: 3.715V

Jul 13 12:50:50 raspberrypi bash[640]:         #37: 3.709V

Jul 13 12:50:50 raspberrypi bash[640]:         #38: 3.711V

Jul 13 12:50:50 raspberrypi bash[640]:         #39: 3.71V

Jul 13 12:50:50 raspberrypi bash[640]:         #40: 3.712V

Jul 13 12:50:50 raspberrypi bash[640]:         #41: 3.712V

Jul 13 12:50:50 raspberrypi bash[640]:         #42: 3.702V

Jul 13 12:50:50 raspberrypi bash[640]:         #43: 3.721V

Jul 13 12:50:50 raspberrypi bash[640]:         #44: 3.719V

Jul 13 12:50:50 raspberrypi bash[640]:         #45: 3.72V

Jul 13 12:50:50 raspberrypi bash[640]:         #46: 3.737V

Jul 13 12:50:50 raspberrypi bash[640]:         #47: 3.738V

Jul 13 12:50:50 raspberrypi bash[640]:         #48: 3.735V

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.264 | DEBUG | Thread-3   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [null]

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.264 | DEBUG | Thread-3   | alyBmsRS485Processor:104   | No bytes received: 2 times!

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.264 | DEBUG | Thread-3   | alyBmsRS485Processor:114   | Waiting for messages to arrive....

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.410 | INFO  | Thread-7   | verter.BmsToInverter:333   | Sending to inverter SOLIS_HV_CAN on can0...

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.410 | DEBUG | Thread-7   | ocol.can.JavaCANPort:76    | CAN frame read...

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.464 | DEBUG | Thread-3   | alyBmsRS485Processor:129   | Invalid frame received! Buffer (HEX): [null]

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.464 | DEBUG | Thread-3   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [null]

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.465 | DEBUG | Thread-3   | alyBmsRS485Processor:104   | No bytes received: 3 times!

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.465 | DEBUG | Thread-3   | alyBmsRS485Processor:114   | Waiting for messages to arrive....

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.665 | DEBUG | Thread-3   | alyBmsRS485Processor:129   | Invalid frame received! Buffer (HEX): [null]

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.665 | DEBUG | Thread-3   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [null]

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.666 | DEBUG | Thread-3   | alyBmsRS485Processor:104   | No bytes received: 4 times!

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13 12:50:50.666 | DEBUG | Thread-3   | alyBmsRS485Processor:114   | Waiting for messages to arrive....

Jul 13 12:50:50 raspberrypi bash[640]: 2025-07-13

ai-republic, 2025-08-29:
Sorry, I've been really busy the last few weeks.
I fixed some issues with the Configurator. Could you please try again?