ai-republic/bms-to-inverter #144: Unable to build

moussa11, 2025-08-14:
OS: RaspianOS 64bit bookworm
java: openjdk 17.0.16 2025-07-15
OpenJDK Runtime Environment (build 17.0.16+8-Debian-1deb12u1)
OpenJDK 64-Bit Server VM (build 17.0.16+8-Debian-1deb12u1, mixed mode, sharing)

Error: 
[INFO] --- compiler:3.10.1:compile (default-compile) @ core-api ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 43 source files to /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/Inverter.java:[26,20] package javax.inject does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[24,32] package javax.enterprise.context does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[25,31] package javax.enterprise.inject does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[26,35] package javax.enterprise.inject.spi does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[28,2] cannot find symbol
  symbol: class ApplicationScoped
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMS.java:[25,20] package javax.inject does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[27,32] package javax.enterprise.context does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[28,31] package javax.enterprise.inject does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[29,35] package javax.enterprise.inject.spi does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[31,2] cannot find symbol
  symbol: class ApplicationScoped
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/protocol/modbus/ModBusPort.java:[15,24] package javax.annotation does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterQualifier.java:[22,20] package javax.inject does not exist
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterQualifier.java:[24,2] cannot find symbol
  symbol: class Qualifier
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/Inverter.java:[34,6] cannot find symbol
  symbol:   class Inject
  location: class com.airepublic.bmstoinverter.core.Inverter
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[53,6] cannot find symbol
  symbol:   class Produces
  location: class com.airepublic.bmstoinverter.core.InverterProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMS.java:[35,6] cannot find symbol
  symbol:   class Inject
  location: class com.airepublic.bmstoinverter.core.BMS
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[51,6] cannot find symbol
  symbol:   class Produces
  location: class com.airepublic.bmstoinverter.core.BMSListProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[57,6] cannot find symbol
  symbol:   class Produces
  location: class com.airepublic.bmstoinverter.core.BMSListProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/protocol/modbus/ModBusPort.java:[46,6] cannot find symbol
  symbol:   class PostConstruct
  location: class com.airepublic.bmstoinverter.core.protocol.modbus.ModBusPort
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[71,24] cannot find symbol
  symbol:   variable CDI
  location: class com.airepublic.bmstoinverter.core.InverterProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[94,25] cannot find symbol
  symbol:   variable CDI
  location: class com.airepublic.bmstoinverter.core.BMSListProducer
[INFO] 21 errors 
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for bms-to-inverter-parent 0.0.1-SNAPSHOT:
[INFO] 
[INFO] bms-to-inverter-parent ............................. SUCCESS [  0.842 s]
[INFO] core-api-0.0.1-SNAPSHOT ............................ FAILURE [  8.701 s]
[INFO] protocol-can-0.0.1-SNAPSHOT ........................ SKIPPED
[INFO] protocol-rs485-0.0.1-SNAPSHOT ...................... SKIPPED
[INFO] bms-daly-common-0.0.1-SNAPSHOT ..................... SKIPPED
[INFO] bms-daly-rs485-0.0.1-SNAPSHOT ...................... SKIPPED
[INFO] inverter-deye-can-0.0.1-SNAPSHOT ................... SKIPPED
[INFO] webserver-0.0.1-SNAPSHOT ........................... SKIPPED
[INFO] bms-to-inverter-main-0.0.1-SNAPSHOT ................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  10.274 s
[... 30 lines cut ...]
[ERROR]   location: class com.airepublic.bmstoinverter.core.BMSListProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[57,6] cannot find symbol
[ERROR]   symbol:   class Produces
[ERROR]   location: class com.airepublic.bmstoinverter.core.BMSListProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/protocol/modbus/ModBusPort.java:[46,6] cannot find symbol
[ERROR]   symbol:   class PostConstruct
[ERROR]   location: class com.airepublic.bmstoinverter.core.protocol.modbus.ModBusPort
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/InverterProducer.java:[71,24] cannot find symbol
[ERROR]   symbol:   variable CDI
[ERROR]   location: class com.airepublic.bmstoinverter.core.InverterProducer
[ERROR] /home/pg/bms-to-inverter/temp/bms-to-inverter-main/core-api/src/main/java/com/airepublic/bmstoinverter/core/BMSListProducer.java:[94,25] cannot find symbol
[ERROR]   symbol:   variable CDI
[ERROR]   location: class com.airepublic.bmstoinverter.core.BMSListProducer
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :core-api

Failed to build application!

ai-republic, 2025-08-17:
My bad, I forgot to push the pom template for the footprint minimization.
It works now 😃

moussa11, 2025-08-17:
Much appreciated. It complied successfully without errors. I ran into two issues when tried to run ./start.sh:

1. the logs had permission write issue. change the ownship to current user (resolved)
2. pg@rpipg:~/bms-to-inverter $ ./start.sh 
2025-08-17 05:50:15.063 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: config/config.properties
Exception in thread "main" java.lang.IllegalStateException: No valid CDI implementation found
	at javax.enterprise.inject.se.SeContainerInitializer.findSeContainerInitializer(SeContainerInitializer.java:106)
	at javax.enterprise.inject.se.SeContainerInitializer.newInstance(SeContainerInitializer.java:97)
	at com.airepublic.bmstoinverter.BmsToInverter.main(BmsToInverter.java:87)

moussa11, 2025-08-17:
Hi, This works now! Perfect. 

I am facing a separate issue. have two Daly RS485 on two UART - USB dongles defined as /dev/ttyUSB0 & /dev/ttyUSB1. its only reading from ttyUSB0 irrespective of which bms or uart dongle connected. 

Any suggestions ?

closed 2025-08-17 by moussa11