ai-republic/bms-to-inverter #147: Support Solax - Pylontech

fritzpille, 2025-08-21:
Hello,
i try to connect a Pylontech HV SC0500-100-V1 and SC1000-100-V2 to a Raspberry Pi with Waveshare CAN Board over bms-to-inverter, but i get no Data over Pylon HV CAN ...

My Solax X3 Hybrid G1/2 works with the SC0500-100-V1, but not with the SC1000-100-V2 ...

I get som information over the solax support, see Attachment, but first the connection to the Pylontech must work ... 😕

[Hybrid X1&X3-G3 ModbusTCP&RTU V3.32- English.pdf](https://github.com/user-attachments/files/21913161/Hybrid.X1.X3-G3.ModbusTCP.RTU.V3.32-.English.pdf)

ai-republic, 2025-08-25:
Can you share some more insights like the config.properties and possibly attach the log file please?

ai-republic, 2025-09-19:
Sorry for the late reply, but I was really busy the last few weeks.
Thanks for the log and configuration. I did see that you had some errors opening the ports - that happens occasionally when the port seems to be allocated. Stopping the port and pulling it up again usually helps, or a reboot.
The communication seems to be ok, but I can't see any response from the battery coming in.

Could you verify that the BMS has the ID 1 set like in your configuration. The ID is often set via a DIP switch on the BMS itself. That has to match the ID you configure in the Configurator. If there is no way to set the ID via DIP switch you can also try setting the ID to 0 in the Configurator.

For testing, could you please set your inverter binding to NONE and also disable the webserver for now and please attach a new log.
Also please delete the BMS-to-Inverter.log before starting the application, then it's easier to view.
Thanks!

GeorgeSasDK, 2025-10-02:
I set up Pylontech on can1 and I can see clean messages from Pylontech but java interpreter sees them as alarms ?
 candump can1
  can1  359   [7]  00 00 00 00 01 50 4E
  can1  351   [8]  14 02 72 01 72 01 C2 01
  can1  355   [4]  59 00 64 00
  can1  356   [6]  83 13 00 00 CE 00
  can1  35C   [2]  C0 00
  can1  35E   [8]  50 59 4C 4F 4E 20 20 20
  can1  359   [7]  00 00 00 00 01 50 4E
  can1  351   [8]  14 02 72 01 72 01 C2 01
  can1  355   [4]  59 00 64 00
  can1  356   [6]  84 13 00 00 CE 00
  can1  35C   [2]  C0 00
  can1  35E   [8]  50 59 4C 4F 4E 20 20 20
  can1  359   [7]  00 00 00 00 01 50 4E
  can1  351   [8]  14 02 72 01 72 01 C2 01
  can1  355   [4]  59 00 64 00
  can1  356   [6]  83 13 00 00 CD 00
  can1  35C   [2]  C0 00
  can1  35E   [8]  50 59 4C 4F 4E 20 20 20
  can1  359   [7]  00 00 00 00 01 50 4E
  can1  351   [8]  14 02 72 01 72 01 C2 01
  can1  355   [4]  59 00 64 00
  can1  356   [6]  84 13 00 00 CD 00
  can1  35C   [2]  C0 00
  can1  35E   [8]  50 59 4C 4F 4E 20 20 20
  can1  359   [7]  00 00 00 00 01 50 4E
  can1  351   [8]  14 02 72 01 72 01 C2 01
  can1  355   [4]  59 00 64 00
  can1  356   [6]  84 13 00 00 CD 00
  can1  35C   [2]  C0 00
  can1  35E   [8]  50 59 4C 4F 4E 20 20 20
  can1  359   [7]  00 00 00 00 01 50 4E
  can1  351   [8]  14 02 72 01 72 01 C2 01
  can1  355   [4]  59 00 64 00
  can1  356   [6]  84 13 00 00 CD 00
  can1  35C   [2]  C0 00
  can1  35E   [8]  50 59 4C 4F 4E 20 20 20

2025-10-02 20:53:04.279 | INFO  | Thread-3   | verter.BmsToInverter:429   | BMS alarms:
        BMS #1: ALARM -> CELL_VOLTAGE_HIGH
        BMS #1: ALARM -> CELL_TEMPERATURE_LOW
        BMS #1: ALARM -> CHARGE_CURRENT_HIGH
        BMS #1: ALARM -> CELL_VOLTAGE_LOW
        BMS #1: ALARM -> DISCHARGE_CURRENT_HIGH

[BMS-to-Inverter.log](https://github.com/user-attachments/files/22666975/BMS-to-Inverter.log)

[CAN-Bus-protocol-low-voltage-V1.3-20190301.pdf](https://github.com/user-attachments/files/22666980/CAN-Bus-protocol-low-voltage-V1.3-20190301.pdf)

I attached the Pylontech CAN documentation also.
I am no programmer but CAN messages from Pylontech seem to be OK ?

0x35E [50 59 4C 4F 4E 20 20 20]

“**PYLON **” → vendor string.

0x359 [00 00 00 00 01 50 4E]

Likely module count = 0x01 → 1 pack

Trailing 50 4E = “PN” (Pylontech marker).
(The four leading 0x00s are reserved/unknown in this frame set.)

0x351 [14 02 72 01 72 01 C2 01]

Interpret as 16-bit little-endian words:

0x0214 = 532 → Max voltage = 53.2 V

0x0172 = 370 → Max charge current = 37.0 A

0x0172 = 370 → Max discharge current = 37.0 A

0x01C2 = 450 → reserved/unknown field (often 0.1-unit; if so that’d be 45.0 of something — not the US3000C nominal Ah, so treat as unknown)

0x355 [59 00 64 00]

0x0059 = 89 → SOC = 89 %

0x0064 = 100 → SOH = 100 %

0x356 (example lines)

83 13 00 00 CE 00 → words: 0x1383, 0x0000, 0x00CE

Voltage = 0x1383 = 4995 → 49.95 V

Current = 0x0000 = 0 → 0.0 A (treat as signed 16-bit /10)

Temp = 0x00CE = 206 → 20.6 °C

You also have 84 13 … CE 00 and … CD 00 → that’s the pack voltage nudging 49.96 V, temp 20.6–20.5 °C, current 0.0 A.

0x35C [C0 00]

Status/alarm bitfield = 0x00C0